espressif/esp-matter #447: Device Attestation Verification Failed

Focalcrest-Madi, 2023-06-07:
I flash the test certificates to the device, but the Attestation Verification failed when the device was paired.
The device and chip-tool log is as follows：
[device.log](https://github.com/espressif/esp-matter/files/11674161/device.log)
[chip-tool.log](https://github.com/espressif/esp-matter/files/11674163/chip-tool.log)

shubhamdp, 2023-06-07:
From `chip-tool.log`
> [1686124594578] [55748:3832836] [CTL] Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors

Please check [here](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/attestation_verifier/DeviceAttestationVerifier.h#L35), `101` means it is unable to find the PAA to verify the attestation certificate chain.

For PAAs not known to chip-tool has to be specified with argument `--paa-trust-store-path <path/to/paa/>`. Please make sure PAA is in der format.

shubhamdp, 2023-06-13:
@Focalcrest-Madi If this worked for you can you please code this issues?

dhrishi, 2023-06-23:
Closing this now

closed 2023-06-23 by dhrishi