espressif/esp-matter #449: ESP32 Dev kit resets on its own after flashing

va1labh, 2023-06-07:
Used ESP-IDF v4.4.2 & a 2nd stage bootloader to build the generic switch example located in the esp-matter repo. 
Error log :
I (0) cpu_start: App cpu up.
I (723) cpu_start: Pro cpu start user code
I (724) cpu_start: cpu freq: 160000000
I (724) cpu_start: Application information:
I (728) cpu_start: Project name:     generic_switch
I (734) cpu_start: App version:      v1.0
I (738) cpu_start: Compile time:     Jun  7 2023 18:54:29
I (744) cpu_start: ELF file SHA256:  6f8fa2f1b3a16234...
I (750) cpu_start: ESP-IDF:          v4.4.2
I (756) heap_init: Initializing. RAM available for dynamic allocation:
I (763) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (769) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (775) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (781) heap_init: At 3FFD3C58 len 0000C3A8 (48 KiB): DRAM
I (787) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (793) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (800) heap_init: At 4009CAC8 len 00003538 (13 KiB): IRAM
I (807) spi_flash: detected chip: generic
I (811) spi_flash: flash io: dio
I (818) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (901) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (901) app_main: Generic Switch created with endpoint_id 1
I (901) gpio: GPIO[6]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
ets Jul 29 2019 12:21:46
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (58) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (58) boot: compile time 18:54:34
I (58) boot: chip revision: 3
I (61) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (68) boot.esp32: SPI Speed      : 40MHz
I (73) boot.esp32: SPI Mode       : DIO
I (77) boot.esp32: SPI Flash Size : 4MB
W (82) boot.esp32: PRO CPU has been reset by WDT.
W (87) boot.esp32: WDT reset info: PRO CPU PC=0x400818e5
0x400818e5: panicHandler at /home/vallabh/MatterPrj/esp-idf/components/esp_system/port/panic_handler.c:213

W (93) boot.esp32: WDT reset info: APP CPU PC=0x400910b0
0x400910b0: _xt_context_save at /home/vallabh/MatterPrj/esp-idf/components/freertos/port/xtensa/xtensa_context.S:209

shubhamdp, 2023-06-09:
@va1labh Can you please share the esp-matter commit you used, and the SoC.

Also, just to confirm does the hello_world from IDF works fine on this?

va1labh, 2023-06-09:
[image]
I have used the ESP32 DevKit V1/ESP-WROOM-32 and the hello-world example works fine.

shubhamdp, 2023-06-13:
Is the IDF v4.4.2 a hard dependency here? Can you use the latest v5.0.1 as mentioned in the docs https://github.com/espressif/esp-matter/#supported-esp-idf-and-connectedhomeip-versions

dhrishi, 2023-06-23:
@va1labh As per the [docs](https://github.com/espressif/esp-matter/#supported-esp-idf-and-connectedhomeip-versions), please use IDF v5.0.1 with the main branch and v4.4.3 with release/v1.0 branch

closed 2023-06-23 by dhrishi