espressif/esp-matter #451: ESPWiFiDriver::OnScanWiFiNetworkDone()

shun-dgut, 2023-06-08:
If I want to call this function ESPWiFiDriver::OnScanWiFiNetworkDone() in the application layer to perform a wifi scan, what should I do?
How should I implement a mpScanCallback->OnFinished function in my application layer?
```
    wifi_ap_record_t * ap_list_buffer = ap_buffer_ptr.get();
    if (esp_wifi_scan_get_ap_records(&ap_number, ap_list_buffer) == ESP_OK)
    {
        if (CHIP_NO_ERROR == DeviceLayer::SystemLayer().ScheduleLambda([ap_number, ap_list_buffer]() {
                std::unique_ptr<wifi_ap_record_t[]> auto_free(ap_list_buffer);
                ESPScanResponseIterator iter(ap_number, ap_list_buffer);
                if (GetInstance().mpScanCallback)
                {
                    GetInstance().mpScanCallback->OnFinished(Status::kSuccess, CharSpan(), &iter);
                    GetInstance().mpScanCallback = nullptr;
                }
                else
                {
                    ChipLogError(DeviceLayer, "can't find the ScanCallback function");
                }
            }))
        {
            ap_buffer_ptr.release();
        }
    }
    else
    {
        ChipLogError(DeviceLayer, "can't get ap_records ");
        if (mpScanCallback)
        {
            mpScanCallback->OnFinished(Status::kUnknownError, CharSpan(), nullptr);
            mpScanCallback = nullptr;
        }
    }
```

shubhamdp, 2023-06-09:
I am not sure what you are trying to do here, so can you elaborate on problem/issue more?

And if I understand correctly, if you just want to scan wifi networks from application, then please take a look at [esp_wifi_scan_start()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv419esp_wifi_scan_startPK18wifi_scan_config_tb). Also please reference this example from IDF https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan

shun-dgut, 2023-06-09:
> I am not sure what you are trying to do here, so can you elaborate on problem/issue more?
>
> And if I understand correctly, if you just want to scan wifi networks from application, then please take a [...]

If I use esp_wifi_scan_start, when the scan completes it will trigger the sweep completion inside void ConnectivityManagerImpl::OnWiFiPlatformEvent(const ChipDeviceEvent * event) and thus the wifi list data will be  read, but I want to get the wifi list in the application layer

wqx6, 2023-06-09:
@shun-dgut so you don't want the ap record to be read by Matter connectivity manager if the esp_wifi_scan_start is not called in Matter NetworkCommissioningDriver? right?

shun-dgut, 2023-06-09:
> @shun-dgut so you don't want the ap record to be read by Matter connectivity manager if the esp_wifi_scan_start is not called in Matter NetworkCommissioningDriver? right?

I want to read the wifi scan list at the application layer, but it will be read by Matter and I can’t read it, but I can’t annotate the read function of matter, because this will make it impossible to pair in the LAN.

wqx6, 2023-06-09:
https://github.com/project-chip/connectedhomeip/pull/27161 this might solve your issue.

closed 2023-06-09 by shun-dgut