espressif/esp-matter #453: Unable to start chip-tool

va1labh, 2023-06-09:
Unable to commission using the chip tool due to the following errors, IDF ver;4.4.2:

`vallabh@bitbeing:~/MatterPrj/esp-matter/examples/generic_switch$ chip-tool interactive start
[1686293448.789559][43808:43808] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /mnt/chip_kvs
[1686293448.789645][43808:43808] CHIP:DL: failed to open file (/mnt/chip_kvs-3yc8jM) for writing
[1686293448.789655][43808:43808] CHIP:-: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/CHIPLinuxStorageIni.cpp:122: Error 0x000000AD at ../examples/chip-tool/commands/common/CHIPCommand.cpp:123
[1686293448.789663][43808:43808] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/CHIPLinuxStorageIni.cpp:122: Error 0x000000AD`

Tried running the `./install` and `. ./export.sh `, but failed to start.

shubhamdp, 2023-06-09:
> CHIP:DL: failed to open file (/mnt/chip_kvs-3yc8jM) for writing

This could be because of permission issues. Usually chip-tool stores its data at `/tmp` not sure why it is using `/mnt`.

Can you check `echo $TMPDIR`, because this is the first thing before falling back to `/tmp`

JFYI, chip-tool and IDF are not related.

va1labh, 2023-06-09:
Its pointing to nowhere
[image]

shubhamdp, 2023-06-12:
Can you try exporting that to somewhere, say `/tmp`?

va1labh, 2023-06-12:
Tried exporting it `export TMPDIR='/home/vallabh/MatterPrj/esp-matter/tmp` but still getting the same error log :

[1686544859.457306][5224:5224] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /mnt/chip_kvs
[1686544859.457370][5224:5224] CHIP:DL: failed to open file (/mnt/chip_kvs-FyP6f4) for writing
[1686544859.457381][5224:5224] CHIP:-: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/CHIPLinuxStorageIni.cpp:122: Error 0x000000AD at ../examples/chip-tool/commands/common/CHIPCommand.cpp:123
[1686544859.457391][5224:5224] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/CHIPLinuxStorageIni.cpp:122: Error 0x000000AD

shubhamdp, 2023-06-13:
@va1labh Can you try without interactive mode? `chip-tool pairing ble-wifi 0x1 ssid password 20202021 3840`

va1labh, 2023-06-13:
Still getting the same error code.

dhrishi, 2023-06-23:
@va1labh This seems to be a chip-tool specific issue. If you are still seeing this issue, can you please report this in `connectedhomeip/issues` for better assistance

closed 2023-06-23 by dhrishi