espressif/esp-matter #454: generic_switch cannot be displayed on apple home

shun-dgut, 2023-06-09:
The code is as follows
```
generic_switch::config_t generic_switch_conf;
bridged_device->endpoint = generic_switch::add(bridged_device->endpoint, &generic_switch_conf);

cluster_t *cluster = cluster::get(bridged_device->endpoint, chip::app::Clusters::Switch::Id);
// cluster::switch_cluster::feature::latching_switch::add(cluster);
cluster::switch_cluster::feature::momentary_switch::add(cluster);
```

jonsmirl, 2023-06-09:
Don't know what is wrong, but I will confirm that generic switch has worked for me in the past on Apple Home. I have not tried it recently.

VaishaliAvhale, 2023-07-13:
The generic_switch was not directly added in Home View, You will find it in your specified room where you have added the generic_switch device.

dhrishi, 2023-07-13:
@shun-dgut We are able to see the generic_switch in Apple Home and also handle the single press event

closed 2023-07-13 by dhrishi