espressif/esp-matter #455: No module named 'matter_idl'

xiaozhengrui, 2023-06-09:
Traceback (most recent call last):
  File "/home/janray/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

xiaozhengrui, 2023-06-09:
[1067/1219] Performing build step for 'chip_gn'
FAILED: esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build esp-idf/chip/lib/libCHIP.a /home/janray/esp-matter/examples/light/build/esp-idf/chip/chip_gn-prefix/src/chip_gn-stamp/chip_gn-build /home/janray/esp-matter/examples/light/build/esp-idf/chip/lib/libCHIP.a
cd /home/janray/esp-matter/examples/light/build/esp-idf/chip && ninja esp32
[1/478] c++ obj/third_party/connectedhomeip/src/credentials/libCredentials.DeviceAttestationCredsProvider.cpp.o
[2/478] c++ obj/third_party/connectedhomeip/src/crypto/libChipCrypto.RandUtils.cpp.o
[3/478] c++ obj/third_party/connectedhomeip/src/credentials/tests/libCertTestVectors.CHIPAttCert_test_vectors.cpp.o
[4/478] c++ obj/third_party/connectedhomeip/src/credentials/tests/libCertTestVectors.CHIPCert_error_test_vectors.cpp.o
[5/478] c++ obj/third_party/connectedhomeip/src/credentials/tests/libCertTestVectors.CHIPCert_unit_test_vectors.cpp.o
[6/478] c++ obj/third_party/connectedhomeip/src/crypto/libChipCrypto.RawKeySessionKeystore.cpp.o
[7/478] c++ obj/third_party/connectedhomeip/src/credentials/libCredentials.DeviceAttestationConstructor.cpp.o
[8/478] c++ obj/third_party/connectedhomeip/src/crypto/libChipCrypto.PersistentStorageOperationalKeystore.cpp.o
[9/478] ACTION //third_party/connectedhomeip/src/controller/data_model:data_model_codegen_codegen(//third_party/connectedhomeip/config/esp32/toolchain:esp32)
FAILED: gen/third_party/connectedhomeip/src/controller/data_model/app/PluginApplicationCallbacks.h gen/third_party/connectedhomeip/src/controller/data_model/app/callback-stub.cpp
python ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../../../../connectedhomeip/connectedhomeip/config/esp32/ --current-path ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model --default-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --current-toolchain=//third_party/connectedhomeip/config/esp32/toolchain:esp32 --capture-output --python-dep-list-files gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen_codegen_metadata_path_list.txt -- ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py --generator cpp-app --output-dir gen/third_party/connectedhomeip/src/controller/data_model --expected-outputs gen/third_party/connectedhomeip/src/controller/data_model/data_model_codegen.expected.outputs ../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/controller/data_model/controller-clusters.matter
Traceback (most recent call last):
  File "/home/janray/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 28, in <module>
    from matter_idl.matter_idl_parser import CreateParser
ModuleNotFoundError: No module named 'matter_idl'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/janray/esp-matter/examples/light/build/esp-idf/chip/../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/codegen.py", line 32, in <module>
    from matter_idl.matter_idl_parser import CreateParser
  File "/home/janray/esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/scripts/py_matter_idl/matter_idl/matter_idl_parser.py", line 7, in <module>
    from lark import Lark
ModuleNotFoundError: No module named 'lark'

shubhamdp, 2023-06-12:
Can you try `source $IDF_PATH/export.sh && python3 -m pip install lark`, since above did not work.

xiaozhengrui, 2023-06-12:
Pip installation of lark will default to the local path of Python and change it to the Python environment under esp idf. I will directly copy lark to this path as follows, and then execute idf.py build to compile successfully

Cp - r~/. local/lib/Python 3.10/site packages/stringcase 1.2.0. dist info~/. espressif/Python_ Env/idf4.4_ Py3.10_ Env/lib/python3.10/site packages

Cp - r~/. local/lib/Python 3.10/site packages/stringcase. py~/. espressif/Python_ Env/idf4.4_ Py3.10_ Env/lib/python3.10/site packages

shubhamdp, 2023-06-12:
> Defaulting to user installation because normal site-packages is not writeable

Above line from your log is trying to say the same that its not able to install lark to idf environment.

Sure, the copying it would have worked. But, I am still not sure, why it did not install it into idf's environment.

Can you please share the output of `which -a python3` after souring the esp-idf/export.sh.

If possible can you try: Pick the Python3 binary from idf's environment, it will look like `/Users/shubhampatil/.espressif/python_env/idf4.4_py3.11_env/bin/python3` and then try installing `lark` with this python.

```
/Users/shubhampatil/.espressif/python_env/idf4.4_py3.11_env/bin/python3 -m pip install lark
```

shubhamdp, 2023-06-13:
@xiaozhengrui If the issue is resolved can you please close this.

closed 2023-06-14 by xiaozhengrui