espressif/esp-matter #457: BLE-Commisionning Issue via OTBR using Chip-Tool

varshilglide, 2023-06-09:
I got an while I am doing BLE Commissioning via OTBR using Chip-Tool at that time BLE Commissioning is working perfectly but after little bit of time it again show me that BLE Commissioning failed. And I am trying this multiple times but this things continues same.

Here are the Logs of Error which I am facing,

sudo ./chip-tool-release pairing ble-thread 5 hex:0e080000000000010000000300000f35060004001fffe002085e07fcf8c068ae1e0708fdd7aea40c32a2fe0510dace045f520854798a902e31f75e5d5b030f4f70656e5468726561642d3863303101028c01041009c84d12f741b345a44811173e4b79390c0402a0f7f8 20202021 3840 --timeout 20000
[1686305757.980432][54176:54176] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-kpMjjI)
[1686305757.980534][54176:54176] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1686305757.980543][54176:54176] CHIP:DL: NVS set: chip-counters/reboot-count = 55 (0x37)
[1686305757.980677][54176:54176] CHIP:DL: Failed to get Ethernet interface
[1686305757.980757][54176:54176] CHIP:DL: Failed to reset Ethernet statistic counts
[1686305757.980815][54176:54176] CHIP:DL: Got WiFi interface: wlp3s0
[1686305757.981417][54176:54176] CHIP:DL: Found the primary WiFi interface:wlp3s0
[1686305757.994416][54176:54176] CHIP:TS: Last Known Good Time: 2023-06-01T07:42:25
[1686305757.994594][54176:54176] CHIP:FP: Fabric index 0x1 was retrieved from storage. Compressed FabricId 0x9216ECB523B332B2, FabricId 0x0000000000000001, NodeId 0x000000000001B669, VendorId 0xFFF1
[1686305757.997321][54176:54176] CHIP:ZCL: Using ZAP configuration...
[1686305757.998769][54176:54176] CHIP:DL: Avahi client registered
[1686305757.999335][54176:54176] CHIP:CTL: Setting attestation nonce to random value
[1686305757.999346][54176:54176] CHIP:CTL: Setting CSR nonce to random value
[1686305757.999663][54176:54178] CHIP:CTL: Setting attestation nonce to random value
[1686305757.999672][54176:54178] CHIP:CTL: Setting CSR nonce to random value
[1686305758.002585][54176:54178] CHIP:CTL: Generating NOC
[1686305758.006036][54176:54178] CHIP:FP: Validating NOC chain
[1686305758.019585][54176:54178] CHIP:FP: NOC chain validation successful
[1686305758.019604][54176:54178] CHIP:FP: Updated fabric at index: 0x1, Node ID: 0x000000000001B669
[1686305758.019609][54176:54178] CHIP:TS: Last Known Good Time: 2023-06-01T07:42:25
[1686305758.019611][54176:54178] CHIP:TS: New proposed Last Known Good Time: 2021-01-01T00:00:00
[1686305758.019613][54176:54178] CHIP:TS: Retaining current Last Known Good Time
[1686305758.019968][54176:54178] CHIP:FP: Metadata for Fabric 0x1 persisted to storage.
[1686305758.020280][54176:54178] CHIP:TS: Committing Last Known Good Time to storage: 2023-06-01T07:42:25
[1686305758.020493][54176:54178] CHIP:CTL: Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 9216ECB523B332B2)
[1686305758.022890][54176:54178] CHIP:CTL: Setting thread operational dataset from parameters
[1686305758.022895][54176:54178] CHIP:CTL: Setting attestation nonce to random value
[1686305758.022900][54176:54178] CHIP:CTL: Setting CSR nonce to random value
[1686305758.022920][54176:54178] CHIP:CTL: Commission called for node ID 0x0000000000000005
[1686305758.028228][54176:54177] CHIP:BLE: BLE removing known devices.
[1686305758.030538][54176:54177] CHIP:BLE: BLE initiating scan.
[1686305758.150052][54176:54178] CHIP:DL: Long dispatch time: 127 ms, for event type 2
[1686305758.173794][54176:54177] CHIP:BLE: New device scanned: FA:A9:80:26:2F:04
[1686305758.173808][54176:54177] CHIP:BLE: Device discriminator match. Attempting to connect.
[1686305759.605568][54176:54178] CHIP:DIS: Closing all BLE connections
[1686305759.605725][54176:54178] CHIP:EM: <<< [E:33218i S:0 M:82920935] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:20 (SecureChannel:PBKDFParamRequest)
[1686305759.605746][54176:54178] CHIP:IN: (U) Sending msg 82920935 to IP address 'BLE'
[1686305760.096763][54176:54178] CHIP:BLE: subscribe complete, ep = 0x555e8541fd00
[1686305760.097114][54176:54178] CHIP:BLE: peripheral chose BTP version 4; central expected between 4 and 4
[1686305760.097129][54176:54178] CHIP:BLE: using BTP fragment sizes rx 244 / tx 244.
[1686305760.097137][54176:54178] CHIP:BLE: local and remote recv window size = 5
[1686305760.235426][54176:54178] CHIP:EM: >>> [E:33218i S:0 M:164785239] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:21 (SecureChannel:PBKDFParamResponse)
[1686305760.255312][54176:54178] CHIP:EM: <<< [E:33218i S:0 M:82920936] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:22 (SecureChannel:PASE_Pake1)
[1686305760.255330][54176:54178] CHIP:IN: (U) Sending msg 82920936 to IP address 'BLE'
[1686305762.125722][54176:54178] CHIP:EM: >>> [E:33218i S:0 M:164785240] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:23 (SecureChannel:PASE_Pake2)
[1686305762.142031][54176:54178] CHIP:EM: <<< [E:33218i S:0 M:82920937] (U) Msg TX to 0:0000000000000000 [0000] --- Type 0000:24 (SecureChannel:PASE_Pake3)
[1686305762.142051][54176:54178] CHIP:IN: (U) Sending msg 82920937 to IP address 'BLE'
[1686305762.257427][54176:54178] CHIP:EM: >>> [E:33218i S:0 M:164785241] (U) Msg RX from 0:0000000000000000 [0000] --- Type 0000:40 (SecureChannel:StatusReport)
[1686305762.257475][54176:54178] CHIP:SC: SecureSession[0x7f9d84009670]: Moving from state 'kEstablishing' --> 'kActive'
[1686305762.257483][54176:54178] CHIP:TOO: Pairing Success
[1686305762.257486][54176:54178] CHIP:TOO: PASE establishment successful
[1686305762.257495][54176:54178] CHIP:CTL: Commissioning stage next step: 'SecurePairing' -> 'ReadCommissioningInfo'
[1686305762.257501][54176:54178] CHIP:CTL: Performing next commissioning step 'ReadCommissioningInfo'
[1686305762.257506][54176:54178] CHIP:CTL: Sending request for commissioning information
[1686305762.257580][54176:54178] CHIP:EM: <<< [E:33219i S:54974 M:68157969] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:02 (IM:ReadRequest)
[1686305762.257591][54176:54178] CHIP:IN: (S) Sending msg 68157969 on secure session with LSID: 54974
[1686305762.527502][54176:54178] CHIP:EM: >>> [E:33219i S:54974 M:258688246] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:05 (IM:ReportData)
[1686305762.528023][54176:54178] CHIP:CTL: ----- NetworkCommissioning Features: has Thread. endpointid = 0
[1686305762.528052][54176:54178] CHIP:CTL: Successfully finished commissioning step 'ReadCommissioningInfo'
[1686305762.528059][54176:54178] CHIP:CTL: Commissioning stage next step: 'ReadCommissioningInfo' -> 'ArmFailSafe'
[1686305762.528064][54176:54178] CHIP:CTL: Performing next commissioning step 'ArmFailSafe'
[1686305762.528082][54176:54178] CHIP:CTL: Arming failsafe (60 seconds)
[1686305762.528159][54176:54178] CHIP:EM: <<< [E:33220i S:54974 M:68157970] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1686305762.528171][54176:54178] CHIP:IN: (S) Sending msg 68157970 on secure session with LSID: 54974
[1686305762.663991][54176:54178] CHIP:EM: >>> [E:33220i S:54974 M:258688247] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[... 84 lines cut ...]
[1686305765.453559][54176:54178] CHIP:TOO: CASE establishment successful
[1686305765.453567][54176:54178] CHIP:CTL: Successfully finished commissioning step 'SendNOC'
[1686305765.453573][54176:54178] CHIP:CTL: No NetworkScan enabled or WiFi/Thread endpoint not specified, skipping ScanNetworks
[1686305765.453579][54176:54178] CHIP:CTL: Commissioning stage next step: 'SendNOC' -> 'ThreadNetworkSetup'
[1686305765.453586][54176:54178] CHIP:CTL: Performing next commissioning step 'ThreadNetworkSetup'
[1686305765.453656][54176:54178] CHIP:EM: <<< [E:33228i S:54974 M:68157978] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1686305765.453670][54176:54178] CHIP:IN: (S) Sending msg 68157978 on secure session with LSID: 54974
[1686305765.633411][54176:54178] CHIP:EM: >>> [E:33228i S:54974 M:258688255] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[1686305765.633533][54176:54178] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1686305765.634211][54176:54178] CHIP:CTL: Received NetworkConfig response, networkingStatus=0
[1686305765.634233][54176:54178] CHIP:CTL: Successfully finished commissioning step 'ThreadNetworkSetup'
[... cut ...]
[1686305765.768757][54176:54178] CHIP:CTL: Performing next commissioning step 'ThreadNetworkEnable'
[1686305765.768814][54176:54178] CHIP:EM: <<< [E:33230i S:54974 M:68157980] (S) Msg TX to 0:FFFFFFFB00000000 [0000] --- Type 0001:08 (IM:InvokeCommandRequest)
[1686305765.768828][54176:54178] CHIP:IN: (S) Sending msg 68157980 on secure session with LSID: 54974
[1686305768.152962][54176:54178] CHIP:BLE: BLE scan error: src/platform/Linux/bluez/ChipDeviceScanner.cpp:154: CHIP Error 0x00000032: Timeout

shubhamdp, 2023-06-09:
@varshilglide Can you please share the below details:
- esp-idf commit
- esp-matter commit
- SoC: (eg: ESP32/C3/H2)
- Attach the logs from the device, please do not paste them in comment, either attach log file or use use details, summary markdown tags.
- Which border router router you are using? Please attach logs if available.

varshilglide, 2023-06-13:
Hey, @shubhamdp Actually the thing is that I am using an NRF52840 Micro-Controller device and Raspberry PI + NRF52840 Dongle as an Open Thread Border Router. I am using Chip-Tool Controller in Linux PC.
Here are the Logs of when I am doing BLE Commissioning between Chip-Tool and OTBR.
[commissioning_log_c.txt](https://github.com/espressif/esp-matter/files/11731200/commissioning_log_c.txt)

shubhamdp, 2023-06-13:
@varshilglide If you are not using an Espressif SoC can please you file the Issue on https://github.com/project-chip/connectedhomeip/issues so that relevant folks can take a look at that and please close this one.

varshilglide, 2023-06-13:
Okay Thank You

closed 2023-06-13 by varshilglide