espressif/esp-matter #459: Get Node ID (Peer ID) in OnAttributeData callback

glennmullerar, 2023-06-12:
Hello,

I'm trying to implement a monitoring system with Matter. 

To retrieve and process data, I subscribe to the attributes I'm interested in. Then I process the values in the ```OnAttributeData``` callback, however I would also like to retrieve the node id of the emitter.

Unfortunately, I can't figure out how to do this with the arguments passed to the function.

Do you have any idea how to do this?

PSONALl, 2023-06-13:
Attribute Path  can be passed as a argument like this:
https://github.com/project-chip/connectedhomeip/blob/6b99adbc3e82095f79822ba497bdb5a5fcd12842/examples/chip-tool/commands/clusters/ReportCommand.h#L34
And you can get node Id from the Attribute path see here: (https://github.com/project-chip/connectedhomeip/blob/6b99adbc3e82095f79822ba497bdb5a5fcd12842/src/app/MessageDef/AttributePathIB.h#L41)

dhrishi, 2023-06-23:
@glennmullerar Please close this if your issue is resolved

glennmullerar, 2023-06-24:
Sorry for not giving any news.

I didn't understand exactly what @PSONALl means.

Should I change the declaration of the ```OnAttributeData``` function to receive ```AttributePathIB``` as a parameter instead of ```ConcreteDataAttributePath```?

Or is there a way to obtain the Node ID from ```ConcreteDataAttributePath```, similar to ```AttributePathIB```?

PS: I haven't found anything like that in the ```connectedhomeip/src/app/ConcreteAttributePath.h``` file.

PSONALl, 2023-07-05:
@glennmullerar You can change the declaration of the OnAttributeData function to receive AttributePathIB 
OR you can use GetNodeId() API from here:  https://github.com/project-chip/connectedhomeip/blob/a9773c943efd9c2dd840bad17b859c31a95ebb3c/src/lib/core/PeerId.h#L37C12-L37C21

dhrishi, 2023-07-20:
You can use the API `GetNodeId` as mentioned above. Please re-open in case of any issue

closed 2023-07-20 by dhrishi