espressif/esp-matter #461: Setting up Python environment error

AyushDubey19, 2023-06-13:
[image]

Error while running ./install.sh

After setting up CIPD package manager successfully error occurred while setting up python environment.

PSONALl, 2023-06-13:
@AyushDubey19 Can you clean build your environment and try again?
1. remove previous environment:
        > cd connectedhomeip/connectedhomeip
        > rm -rf .environment
2. Build the environment again using ./install.sh

shubhamdp, 2023-06-13:
@AyushDubey19 Have you installed the prerequisites mentioned in the documentation guide [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup)?

AyushDubey19, 2023-06-13:
> @AyushDubey19 Can you clean build your environment and try again?
> 
> 1. remove previous environment:
>    > cd connectedhomeip/connectedhomeip
>    > rm -rf .environment
> 2. Build the environment again using ./install.sh

Already tried. Still getting same error.

AyushDubey19, 2023-06-13:
@PSONALl  
Already tired. But still getting same error.

AyushDubey19, 2023-06-13:
@shubhamdp I Only completed prerequisites for idf.

shubhamdp, 2023-06-13:
@AyushDubey19 Please install the prerequisites for Matter as described in the documentation.

AyushDubey19, 2023-06-13:
@shubhamdp 

Having issue while running prerequisites command 
"sudo apt-get install git gcc g++ pkg-config libssl-dev libdbus-1-dev \
     libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
     python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev" 

[image]

AyushDubey19, 2023-06-13:
@shubhamdp Thanks Shubham. After install the prerequisites for matter error gone.

DrakoPOD, 2024-05-23:
@shubhamdp My problem is that I using OpenSuse Tumbleweed, there are many dependencies that cannot be found. Also, can I use anaconda enviroments instead of default python?