espressif/esp-matter #465: BLE Commissioning Issue in Chip-Tool

varshilglide, 2023-06-13:
Logs of Chip-To
[logs.txt](https://github.com/espressif/esp-matter/files/11733887/logs.txt)

wqx6, 2023-06-13:
Looks like the chip-tool cannot resolve the address of the operational node. Is the device a Thread device or Wi-Fi device? Is the device(or the Thread BR) in the same local network with the chip-tool?

varshilglide, 2023-06-13:
We are using Open Thread Border Router in Raspberry PI and the devices are thread devices and they are connected to my company's Wi-Fi.

wqx6, 2023-06-14:
So, you are pairing the thread device, maybe this commands could help you
```
sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra=2
sudo sysctl -w net.ipv6.conf.enp2s0.accept_ra_rt_info_max_plen=128
sudo sysctl -w net.ipv6.conf.all.forwarding=1
```
Note: Please replace the `enp2s0` with your network interface name

varshilglide, 2023-06-14:
Yes, we are pairing the threading device.
At which place I have to run above commands that you given.

wqx6, 2023-06-15:
You should run these commands on the chip-tool terminal before you pair the device with chip-tool.

varshilglide, 2023-06-15:
yes, @wqx6 I applied the steps you gave but still not changes anything in the outcome.
Still, as earlier, same BLE Commissioning issues I got.

wqx6, 2023-06-16:
Is the backbone netif of the OTBR connected to the same local network of the chip-tool? could you please provide the log from the OTBR and the device? The log shows that the chip-tool cannot resolve the address of the thread device. The reason might be that the device fails to register SRP service on the SRP server of the BR.

varshilglide, 2023-06-16:
Yes, OTBR is connected to the same local network as the Chip Tool.
[Chip Tool Logs using Latest Version of Chip-Tool.txt](https://github.com/espressif/esp-matter/files/11766498/Chip.Tool.Logs.using.Latest.Version.of.Chip-Tool.txt)
[Chip Tool Logs.txt](https://github.com/espressif/esp-matter/files/11766510/Chip.Tool.Logs.txt)

wqx6, 2023-06-16:
Could you please provide us the thread device-log? 
Also I notice that the chip-tool are using the minimal mDNS, you can rebuild the chip-tool with the platform mDNS. 
```
cd connectedhomeip/examples/chip-tool
gn gen out --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false'
ninja -C out
```

dhrishi, 2023-06-23:
@varshilglide Please provide the requested information or close the issue if it is resolved

varshilglide, 2023-06-26:
Okay Thanks

closed 2023-06-26 by varshilglide