espressif/esp-matter #466: Open commissioning window error

JonathanWitthoeft, 2023-06-13:
```
esp_err_t app_matter_open_commissioning(void)
{
    chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
    constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(k_timeout_seconds);
    if (!commissionMgr.IsCommissioningWindowOpen())
    {
        CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds,
                                        chip::CommissioningWindowAdvertisement::kDnssdOnly);
        if (err != CHIP_NO_ERROR)
        {
        	return ESP_FAIL;
        }
    }
    return ESP_OK;
}
```

I am calling the above code in 2 places. The first one is from the matter callback when all fabrics are removed. That one works and opens on network commissioning. 
The second one is triggered by a button press in my own thread. This one doesn't work and I get the following error logs:
```
E (84728) chip[DL]: Chip stack locking error at 'src/system/SystemLayerImplFreeRTOS.cpp:55'. Code is unsafe/racy
E (84728) chip[-]: chipDie chipDie chipDie
abort() was called at PC 0x40139b7f on core 0
0x40139b7f: chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonathan/Desktop/Workspace/Rollease-Matter-Bridge/firmware/build/esp-idf/chip/../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:508
 (inlined by) chipDie at /home/jonathan/Desktop/Workspace/Rollease-Matter-Bridge/firmware/build/esp-idf/chip/../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/lib/support/CodeUtils.h:518
 (inlined by) chip::Platform::Internal::AssertChipStackLockedByCurrentThread(char const*, int) at /home/jonathan/Desktop/Workspace/Rollease-Matter-Bridge/firmware/build/esp-idf/chip/../../../../esp-matter/connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/LockTracker.cpp:36
```

Is there a safe way to open a commissioning window from my own task, that will not trigger a panic?

shubhamdp, 2023-06-13:
When interacting with matter resources, eg: get/set attributes, invoking any command, it has to be done from matter stack, otherwise above crash happens.
If you are accessing any matter resource please call it between lock/unlock calls

```
chip::DeviceLayer::PlatformMgr().LockChipStack();
.... // access any matter attribute, open/close commissioning window.
chip::DeviceLayer::PlatformMgr().UnlockChipStack();
```

You have to call OpenBasicCommissioningWindow() with by locking chip stack.

shubhamdp, 2023-06-14:
Many people asks this question, I think I'll move it to FAQs section :-D

JonathanWitthoeft, 2023-06-14:
> When interacting with matter resources, eg: get/set attributes, invoking any command, it has to be done from matter stack, otherwise above crash happens. If you are accessing any matter resource please call it between lock/unlock calls

In the app_driver source files for the examples, I do not see this lock being used when updating attributes:
https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_driver.cpp#LL70C16-L70C16

@shubhamdp, can you clarify when it is needed and when it is not?

JonathanWitthoeft, 2023-06-14:
I have adjusted my code to use the chip_stack_lock and unlock functions in `esp_matter_core.h` and it seems to be working. I have also put these locks around all attribute get/sets. Thanks for the feedback!

```
esp_err_t app_matter_open_commissioning(void)
{
	esp_err_t status = ESP_FAIL;

	lock::status_t lock_status = lock::chip_stack_lock(portMAX_DELAY);
	if (lock_status == lock::FAILED) {
		ESP_LOGE(TAG, "Could not lock chip stack");
		return ESP_FAIL;
	}

	chip::CommissioningWindowManager & commissionMgr = chip::Server::GetInstance().GetCommissioningWindowManager();
	constexpr auto kTimeoutSeconds = chip::System::Clock::Seconds16(k_timeout_seconds);
	if (!commissionMgr.IsCommissioningWindowOpen())
	{
		CHIP_ERROR err = commissionMgr.OpenBasicCommissioningWindow(kTimeoutSeconds,
										chip::CommissioningWindowAdvertisement::kDnssdOnly);
		if (err != CHIP_NO_ERROR)
		{
			ESP_LOGE(TAG, "Failed to open commissioning window, err:%" CHIP_ERROR_FORMAT, err.Format());
		}
		else{
			status = ESP_OK;
			ESP_LOGD(TAG, "Commissioning should have opened ");
		}
	}

	if (lock_status == lock::SUCCESS) {
		lock::chip_stack_unlock();
	}

	return status;
}
```

closed 2023-06-14 by JonathanWitthoeft

aggaddam, 2023-09-04:
I am also facing the same issue. Do I need to perform the locking/unlocking operation before accessing any chip stack resource in the matter callback ?  The reason why i am asking is lock/unlocking works perfectly fine if it called from any application thread/task but if we try to perform locking/unlocking in matter app callback, I see that device crashes sometimes.