espressif/esp-matter #467: Commisioning over Soft AP

JonathanWitthoeft, 2023-06-13:
Continuation of https://github.com/espressif/esp-matter/issues/109

Matter v1.1 is released. How can we enable Soft-AP provisioning instead of BLE provisioning for the esp32?

jonsmirl, 2023-06-19:
Provisioning using what? IOS and Google? Is that ever going to work? They'd have to support Soft-AP in their apps.

Provisioning using using your own app? You can do that right now. Use your custom app code with Soft-AP to get the node up on wifi, then do standard IP commissioning (Ethernet commissioning).

jonsmirl, 2023-06-19:
Doesn't look like they have implemented it.  Only a single reference in the source code.

jonsmirl@ares:~/aosp/esp-matter/components$ grep -r PAIRING_MODE_SOFTAP *
esp_matter_controller/esp_matter_controller_pairing_command.h:    PAIRING_MODE_SOFTAP,

Did you consider amending your FCC filing to enable BLE?

jonsmirl, 2023-06-19:
Feedback I have one soft-ap is that customers hate it. They don't understand when phone is in 5Ghz mode and device is at 2.4Ghz.

JonathanWitthoeft, 2023-06-19:
> Did you consider amending your FCC filing to enable BLE?

Yes unfortunately they will not amend it and it would need a full new certification with new FCC ID. Enabling a wireless interface is not small enough to be a permissible change. 
>They don't understand when phone is in 5Ghz mode and device is at 2.4Ghz

BLE and Soft-AP commissioning should act the same with regards to the 5Ghz issue. It all depends on the Wi-Fi band the phone is on when commissioning starts. Matter actually has devices specify the bands they support in the Network Commissioning Cluster so I think the 5Ghz incompatibility will be caught early on during commissioning and an error should be displayed by Google/Apple etc.

jonsmirl, 2023-06-19:
BLE does not experience the 5Ghz issue. BLE only works on 2.4Ghz. Once the device is commissioned it attaches to the router at 2.4GHz. The phone is attached to the router at 5Ghz. The router translates and the customer is unaware.

JonathanWitthoeft, 2023-06-19:
> BLE does not experience the 5Ghz issue. BLE only works on 2.4Ghz. Once the device is commissioned it attaches to the router at 2.4GHz. The phone is attached to the router at 5Ghz. The router translates and the customer is unaware.

But then how does it know the proper SSID to connect to the router with? Is this with the assumption that both 5Ghz and 2.4Ghz of the desired network have the same SSID and Password?

jonsmirl, 2023-06-19:
Same SSID/PW on 2.4Ghz and 5Ghz is preferred because your phone will measure both signal strengths and connect using the better one. I can see this walking around my house. Near the routers it will use 5Ghz. If I go outside onto the deck it will drop down to 2.4Ghz which has better signal propagation.

shubhamdp, 2023-06-21:
@JonathanWitthoeft If you have devices in field, then they will be already connected to the WiFi right? In that case nothing extra is required in the firmware and device will be commissioned using on-network.

There may be some work required after factory reset. But, you may have some flow already present right? So, use the vendor app and commission the device to WiFi network and then on-network commissioning. (This is non-standard commissioning flow, as documented in the specification)

JonathanWitthoeft, 2023-06-21:
@shubhamdp I appreciate the suggestions, but new devices will also be produced, and there will be instances where devices are factory reset. We can not assume all devices will already be commissioned to a network. Part of the appeal of Matter is that you don't require a third party/manufacturer app. 

I guess I would just like to know if the Soft-AP commissioning according to the Matter specification is implemented in the esp-matter SDK, or if there are plans to implement it. From what I can tell it is now implemented in the upstream connectedhomeip repository and it could be as simple as setting some config values. 

From the specification it looks like it is just a Soft-AP SSID format "MATTER-ddd-vvvv-pppp" and a flag or 2 in the network commissioning cluster to show it is supported. 

I will attempt to implement it myself and offer a push request if successful

JonathanWitthoeft, 2023-06-21:
Matter Soft AP creation in ConnectivityManager:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp#L933

Matter commit regarding Soft-AP rendezvous mode:
https://github.com/project-chip/connectedhomeip/commit/5f94d380c22c46267c521a008aff36653b902ecb

shubhamdp, 2023-06-21:
Current code only starts AP and communication after that is not there. Soft AP commissioning would require adding support for Matter Vendor-specific IE, that is not there yet.

And the commit you shared... both soft ap and wifi code was getting compiled and soft ap related code is not required if you have BLE, so this commit just disabled the softap feature at compile time.

dhrishi, 2023-06-23:
@JonathanWitthoeft Soft-AP based commissioning, even though present in the spec, is not fully supported in the upstream connectedhomeip since Matter v1.0 release. There are no chances of supporting this in the subsequent releases too. AFAIK, there's a Wi-Fi effort going on to deprecate Soft-AP and instead use some other modern approach. At this moment, I am not sure of the timelines about it getting into the spec version.

I will close this as the question is answered.