espressif/esp-matter #468: OnMessageReceived failed, err = 70

shun-dgut, 2023-06-14:
```
CONFIG_MAX_EXCHANGE_CONTEXTS=8
CONFIG_MAX_BINDINGS=8
CONFIG_MAX_FABRICS=5
CONFIG_MAX_PEER_NODES=16
CONFIG_MAX_UNSOLICITED_MESSAGE_HANDLERS=8
```
How should these parameters be set to have better performance.
My code comes up with
```
E (806945) chip[EM]: OnMessageReceived failed, err = 70
```

shripad621git, 2023-06-14:
OnMessageReceived failed, err = 70 stands for CHIP_ERROR_UNSOLICITED_MESSAGE_NO_ORIGINATOR. It occurs when the unsolicited message i.e the message that does not match any existing exchange, is not marked as being sent by an initiator.
The message header is checked for an initiator, and if not found the message is dropped and the error above is thrown just for indicating the above stated reason.

jonsmirl, 2023-06-19:
OnMessageReceived failed, err = 70.  They are corrupted packets of some type. They don't hurt anything.
Maybe you can track down where they are coming from.

Why do you think performance is not good?  Thread is slow, but that's because it is Thread.

dhrishi, 2023-06-23:
@shun-dgut Closing this as the analysis of the error statement is mentioned by @shripad621git . Please let us know if you see any issues around this.

closed 2023-06-23 by dhrishi

jonsmirl, 2023-10-01:
This finally got fixed in CHIP.

https://github.com/project-chip/connectedhomeip/pull/29166