espressif/esp-matter #470: How to get Google Home working as CSA Member with company vendor ID

mgoeller, 2023-06-14:
What we have:
-our company vendor ID 0x1...
-we are an official CSA Adopter
-Espressif esp-matter based firmware, compiled with test vendor ID 0xFFF1
-works with Apple Home as un-certified
-works with Google Home (after registering the device in the Matter console with a test vendor ID 0xFFF1
-filled out and sent DCL vendor enrolment form to CSA

What we want:
-test our firmware with our own company vendor ID 0x1... before we ship to “test house” / certification lab

Steps to reproduce:
If we built our firmware using our company vendor ID 0x1... with a custom created CD file, PAA key and PAA certificate, we cannot pair with Google Home.

Questions:
1. What do we need to do? Do we need to register on DCL main net or on DCL test net?
2. Do I understand correctly that we only need main net when we are officially certified?
3. For self created PAA certificates, do we need to use the DCL test net?

chshu, 2023-06-16:
@mgoeller In order to test with Goolge using test DAC, you need to register the device  info in the Google develop console: https://developers.home.google.com/matter/get-started

[image]

> What do we need to do? Do we need to register on DCL main net or on DCL test net?

Only certificied device can be added to DCL mainnet, and you need to register the devcice to Google develop console for other cases.

> Do I understand correctly that we only need main net when we are officially certified?

Yes

> For self created PAA certificates, do we need to use the DCL test net?

You need to check with the test lab, they may ask you to upload the test PAA to the test net before certification test. But anyway you need to share the test PAA with the testlab.

You may also refer to the link for Matter certification FAQs: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/certification.html

jonsmirl, 2023-06-19:
Also, look at the logs from your Android phone. It will usually say why it is failing.

dhrishi, 2023-06-23:
@mgoeller Please close the issue if your questions are answered.

closed 2023-06-29 by dhrishi