espressif/esp-matter #471: How can I add new cluster support in the matter esp controller example ?

amprazak, 2023-06-15:


amprazak, 2023-06-15:
I got this response from the controller when I try to read my cluster attribute : 

> I (319761) chip[TOO]: Endpoint: 0 Cluster: 0x131B_FC00 Attribute 0x0000_0000 DataVersion: 1791934710
> I (319771) chip[TOO]:   Don't know how to log atribute value

jadhavrohit924, 2023-06-15:
@amprazak Can you please explain a bit more, what are you trying to achieve?

> I (319761) chip[TOO]: Endpoint: 0 Cluster: 0x131B_FC00 Attribute 0x0000_0000 DataVersion: 1791934710
I (319771) chip[TOO]: Don't know how to log atribute value

This looks like chip-tool log and when you add new cluster you have to create an cluster-name.xml and re-generate zap. Zap regeneration will auto gen controller code.

amprazak, 2023-06-16:
Yes of course. 
I added a new cluster, I tested it with chip-tool on Linux target everything works perfectly. Now I try to use the [esp controller](https://github.com/espressif/esp-matter/tree/main/examples/controller) to control the new cluster. I don’t know which file to modify to support my new cluster.

wqx6, 2023-06-16:
`chip-tool` will also return `Don't know how to log atribute value` like our controller example, but it enables PRETTY_PRINT by default. If you enable PRETTY_PRINT on our controller. it should also be able to decode and print the attributes on custom cluster.

amprazak, 2023-06-16:
Nice. 
@wqx6  How can I enable PRETY_PRINT on the controller?

wqx6, 2023-06-16:
You should set `enable_im_pretty_print` to true in `args.gni` for gn building and enable `chip_detail_logging` by changing `CONFIG_LOG_DEFAULT_LEVEL` to `DEBUG` in menuconfig

amprazak, 2023-06-16:
@wqx6 thank you. 
You save my day

closed 2023-06-16 by amprazak