espressif/esp-matter #473: This accessory is not responding on HomeKit based on light example

law-ko, 2023-06-17:
Hello,

In the main branch using the `light` example, when pairing up with HomeKit it shows `This accessory is not responding`.

[image]

I have tried with ESP32-PICO-MINI-02, ESP32, and ESP32C3 all had the same issue.

shubhamdp, 2023-06-19:
@law-ko Please share the logs from device, esp-idf and esp-matter commit IDs.

shubhamdp, 2023-06-19:
Also, can you try flashing light example on C3 from here: https://espressif.github.io/esp-launchpad/?solution=matter

law-ko, 2023-06-19:
> @law-ko Please share the logs from device, esp-idf and esp-matter commit IDs.

@shubhamdp 

ESP-IDF commit ID `a4afa44435ef4488d018399e1de50ad2ee964be8` HEAD V5.0.1
ESP-Matter commit ID `f8105768252e89dd6bd5b5bb9b0f3f2118b0edff (HEAD -> main, origin/main, origin/HEAD)`

The launchpad wasn't able to connect to my ESP32.

shubhamdp, 2023-06-19:
It only supports, Google Chrome, were you using Chrome?

Also, please share the logs from device for earlier run.

law-ko, 2023-06-19:
@shubhamdp I am trying on Chrome with Mac and the serial port cannot be found. I've also used the same commit ID to test both ESP32 and ESP32C3 they both failed.

law-ko, 2023-06-19:
@shubhamdp  Here is the datalog, there is no further output after pairing with HomeKit, this is on ESP32C3 with above commit ID
```
I (30) boot: ESP-IDF v5.0.1 2nd stage bootloader
I (30) boot: compile time 23:12:18
I (30) boot: chip revision: v0.3
I (32) boot.esp32c3: SPI Speed      : 80MHz
I (37) boot.esp32c3: SPI Mode       : DIO
I (42) boot.esp32c3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (70) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (78) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (85) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (93) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (100) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (108) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (115) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (123) boot: End of partition table
I (127) boot: No factory image, trying OTA 0
I (132) esp_image: segment 0: paddr=00020020 vaddr=3c100020 size=373b0h (226224) map
I (174) esp_image: segment 1: paddr=000573d8 vaddr=3fc94200 size=03164h ( 12644) load
I (177) esp_image: segment 2: paddr=0005a544 vaddr=40380000 size=05ad4h ( 23252) load
I (184) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=feb28h (1043240) map
I (345) esp_image: segment 4: paddr=0015eb50 vaddr=40385ad4 size=0e640h ( 58944) load
I (362) boot: Loaded app from partition at offset 0x20000
I (406) boot: Set actual ota_seq=1 in otadata[0]
I (406) boot: Disabling RNG early entropy source...
I (417) cpu_start: Pro cpu up.
I (427) cpu_start: Pro cpu start user code
I (427) cpu_start: cpu freq: 160000000 Hz
I (427) cpu_start: Application information:
I (430) cpu_start: Project name:     light
I (435) cpu_start: App version:      v1.0
I (439) cpu_start: Compile time:     Jun 18 2023 23:12:08
I (445) cpu_start: ELF file SHA256:  75dfa3c687d89969...
I (451) cpu_start: ESP-IDF:          v5.0.1
I (456) cpu_start: Min chip rev:     v0.3
I (461) cpu_start: Max chip rev:     v0.99 
I (466) cpu_start: Chip rev:         v0.3
I (470) heap_init: Initializing. RAM available for dynamic allocation:
I (478) heap_init: At 3FCA8C30 len 00033AE0 (206 KiB): DRAM
I (484) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (491) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (498) spi_flash: detected chip: generic
I (502) spi_flash: flash io: dio
W (506) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (517) coexist: coexist rom version 9387209
I (521) cpu_start: Starting scheduler.
I (530) led_driver_ws2812: Initializing light driver
I (530) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
W (550) esp_matter_core: Command 0x00000047 on cluster 0x00000300 already exists. Not creating again.
I (550) app_main: Light created with endpoint_id 1
I (560) pp: pp rom version: 9387209
I (560) net80211: net80211 rom version: 9387209
I (580) wifi:wifi driver task: 3fcb4670, prio:23, stack:6656, core=0
I (580) system_api: Base MAC address is not set
I (580) system_api: read default base MAC address from EFUSE
I (580) wifi:wifi firmware version: 17afb16
I (590) wifi:wifi certification version: v7.0
I (590) wifi:config NVS flash: enabled
I (590) wifi:config nano formating: disabled
I (600) wifi:Init data frame dynamic rx buffer num: 32
I (600) wifi:Init management frame dynamic rx buffer num: 32
I (610) wifi:Init management short buffer num: 32
I (610) wifi:Init dynamic tx buffer num: 32
I (620) wifi:Init static tx FG buffer num: 2
I (620) wifi:Init static rx buffer size: 1600
I (620) wifi:Init static rx buffer num: 10
[... 737 lines cut ...]
I (44630) chip[EM]: >>> [E:56415r S:0 M:156920698 (Ack:219481739)] (U) Msg RX from 0:766752F722F26FD1 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (44640) app_main: Fabric is committed
I (44650) chip[SVR]: Commissioning completed successfully
I (44650) chip[DIS]: Updating services using commissioning mode 0
I (44670) chip[DIS]: CHIP minimal mDNS started advertising.
I (44680) chip[DIS]: Advertise operational node 638C92F1B94D5355-000000004974C1F1
I (44680) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 638C92F1B94D5355-000000004974C1F1.
I (44700) chip[DIS]: mDNS service published: _matter._tcp
I (44700) chip[DIS]: Advertise operational node 0459C5EAC7ADA78D-000000004AD8F3A3
I (44710) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: 0459C5EAC7ADA78D-000000004AD8F3A3.
I (44730) chip[DIS]: mDNS service published: _matter._tcp
I (44730) chip[SC]: SecureSession[0x3fc9b480]: Moving from state 'kActive' --> 'kPendingEviction'
I (44740) app_main: Commissioning complete
I (44750) esp_matter_core: Commissioning Complete
I (44750) chip[EM]: >>> [E:56416r S:60367 M:174668873 (Ack:13365380)] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (44770) chip[EM]: >>> [E:56417r S:60367 M:174668874] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0001:02 (IM:ReadRequest)
I (44770) app_main: Commissioning window closed
[... cut ...]
I (44830) chip[EM]: >>> [E:56418r S:60367 M:174668876] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0001:02 (IM:ReadRequest)
I (44840) chip[EM]: <<< [E:56418r S:60367 M:13365382 (Ack:174668876)] (S) Msg TX to 2:491622BAC79E9642 [A78D] --- Type 0001:05 (IM:ReportData)
I (44850) chip[IN]: (S) Sending msg 13365382 on secure session with LSID: 60367
I (44870) chip[EM]: >>> [E:56418r S:60367 M:174668877 (Ack:13365382)] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

marciogranzotto, 2023-06-19:
I'm having the same issue. Also tried adding it to SmartThings, using the SmartThings Station (Thread), but it gets to the same logs as HomeKit, being already connected to the Thread network but not responding to Matter commands.

I'm using the ESP32-C6 with a modified `sdkconfig.defaults.esp32c6`:
```
CONFIG_IDF_TARGET="esp32c6"

# libsodium
CONFIG_LIBSODIUM_USE_MBEDTLS_SHA=y

# NIMBLE
CONFIG_BT_ENABLED=y
CONFIG_BT_NIMBLE_ENABLED=y
CONFIG_BT_NIMBLE_EXT_ADV=n
CONFIG_BT_NIMBLE_HCI_EVT_BUF_SIZE=70
CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n

# FreeRTOS should use legacy API
CONFIG_FREERTOS_ENABLE_BACKWARD_COMPATIBILITY=y

# Enable OpenThread
CONFIG_OPENTHREAD_ENABLED=y
CONFIG_OPENTHREAD_SRP_CLIENT=y
CONFIG_OPENTHREAD_DNS_CLIENT=y
CONFIG_OPENTHREAD_LOG_LEVEL_DYNAMIC=n
CONFIG_OPENTHREAD_LOG_LEVEL_NOTE=y
CONFIG_OPENTHREAD_CLI=n

# Disable lwip ipv6 autoconfig
CONFIG_LWIP_IPV6_AUTOCONFIG=n

# Use a custom partition table
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_thread.csv"

# LwIP config for OpenThread
CONFIG_LWIP_IPV6_NUM_ADDRESSES=8
CONFIG_LWIP_MULTICAST_PING=y

# MDNS platform
CONFIG_USE_MINIMAL_MDNS=n
CONFIG_ENABLE_EXTENDED_DISCOVERY=y

# Enable OTA Requestor
CONFIG_ENABLE_OTA_REQUESTOR=y

# Disable STA and AP for ESP32C6
CONFIG_ENABLE_WIFI_STATION=n
CONFIG_ENABLE_WIFI_AP=n

# Button
CONFIG_BUTTON_PERIOD_TIME_MS=20
CONFIG_BUTTON_LONG_PRESS_TIME_MS=5000

# Enable chip shell
CONFIG_ENABLE_CHIP_SHELL=y

# Enable DS Peripheral
CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=y
```

esp-matter: bd692f3b9e600d93fc3a70400b57db1e6d6d9902
esp-idf: ea5e0ff298e6257b31d8e0c81435e6d3937f04c7

shubhamdp, 2023-06-20:
@law-ko Logs are pretty clear, Apple Home added the device to both of its fabric and then see the following logs.
```
I (44830) chip[EM]: >>> [E:56418r S:60367 M:174668876] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0001:02 (IM:ReadRequest)
I (44840) chip[EM]: <<< [E:56418r S:60367 M:13365382 (Ack:174668876)] (S) Msg TX to 2:491622BAC79E9642 [A78D] --- Type 0001:05 (IM:ReportData)
I (44850) chip[IN]: (S) Sending msg 13365382 on secure session with LSID: 60367
I (44870) chip[EM]: >>> [E:56418r S:60367 M:174668877 (Ack:13365382)] (S) Msg RX from 2:491622BAC79E9642 [A78D] --- Type 0000:10 (SecureChannel:StandaloneAck)
```

`>>>` logs are data received to the device and `<<<` are data sent from device.

See the last few logs, device received read request and it served it with response and got an ack.

So, device is not stuck, its not receiving any request from the client. Can you check if you see the same behaviour using chip-tool? Commission using chip-tool and try reading few attributes, and you do a wildcard subscribe.
```
./chip-tool any subscribe-by-id 0xFFFFFFFF 0xFFFFFFFF 3 10 (NODE_ID) (EP_ID)
```

Can you please try that and attach the log file (please don't paste it here) for both esp32 and chip-tool?

marciogranzotto, 2023-06-26:
It seems that my issue was something I'd done to the config variables. Deleting `sdkconfig` and making a clean configuration solved the problem.

dhrishi, 2023-06-28:
Thanks for the confirmation @marciogranzotto . @law-ko I will close this issue now. Please re-open if you still see it, with the data requested

closed 2023-06-28 by dhrishi