espressif/esp-matter #476: How to generate QR code for matter devices ? Is there any tool to generate QR code image ?

aggaddam, 2023-06-21:
I want to generate a QR code for the matter device. By default, all the device's discriminator values are 3840.  I am overwriting default nsv data partition with the factory partition data with discriminator value 3001.  I also want to generate a QR code and the new factory partition. Is there a tool we can use to generate QR code ?

Generating factory partition table with Discriminator 3001

cd esp-idf
source export.sh

cd esp-matter
source export.sh

The script generates factory_partition.bin. 
python3 connectedhomeip/connectedhomeip/scripts/tools/generate_esp32_chip_factory_bin.py -d 3001

Flash it on the device.
esptool.py -p (PORT) write_flash 0x10000 factory_partition.bin

```
Version:             0
Vendor ID:           65521
ProductID:           32768
Custom flow:         0        (STANDARD)
Discovery Bitmask:   0x02     (BLE)
Long discriminator:  3001 
Passcode:            20202021
```

shubhamdp, 2023-06-21:
We already documented this in the FAQ...

https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-10-how-to-generate-matter-onboarding-codes-qr-code-and-manual-pairing-code

shubhamdp, 2023-06-23:
@aggaddam Can you please close the issue, if above comment answered you question!

closed 2023-06-23 by dhrishi