espressif/esp-matter #479: 【Light example 】 How to test RGB Feature ?

zsr391912829, 2023-06-25:
I use AiDot APP(Android) to test Light example
[on/off]  OK  
[set level] OK
[set temperature color] OK
[set rgb color] fail

when I user app to set rgb color ,  i found esp32 can't log.
 **[set rgb color LOG]** **FAIL**
> I (396673) chip[EM]: >>> [E:57307r S:21328 M:240716379] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0001:08 (IM:InvokeCommandRequest)
I (396683) chip[EM]: <<< [E:57307r S:21328 M:64842710 (Ack:240716379)] (S) Msg TX to 1:000000000001B669 [193C] --- Type 0001:09 (IM:InvokeCommandResponse)
I (396703) chip[IN]: (S) Sending msg 64842710 on secure session with LSID: 21328
I (396703) chip[EM]: >>> [E:57308r S:21328 M:240716380] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0001:08 (IM:InvokeCommandRequest)
I (396723) chip[EM]: <<< [E:57308r S:21328 M:64842711 (Ack:240716380)] (S) Msg TX to 1:000000000001B669 [193C] --- Type 0001:09 (IM:InvokeCommandResponse)
I (396733) chip[IN]: (S) Sending msg 64842711 on secure session with LSID: 21328
I (396743) chip[EM]: >>> [E:57307r S:21328 M:240716381 (Ack:64842710)] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (396763) chip[EM]: >>> [E:57308r S:21328 M:240716382 (Ack:64842711)] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0000:10 (SecureChannel:StandaloneAck)

[set temperature color LOG ] OK
) chip[IN]: (S) Sending msg 64842725 on secure session with LSID: 21328
I (465283) chip[EM]: >>> [E:57309r S:21328 M:240716390] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0001:08 (IM:InvokeCommandRequest)
I (465293) esp_matter_command: Received command 0x00000000 for endpoint 0x0001's cluster 0x00000008
I (465293) chip[ZCL]: RX level-control: MOVE_TO_LEVEL db  0 0 0
I (465303) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x0000000F is 0 **********
I (465323) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000006's Attribute 0x00000000 is 1 **********
I (465333) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (465343) chip[EM]: <<< [E:57309r S:21328 M:64842726 (Ack:240716390)] (S) Msg TX to 1:000000000001B669 [193C] --- Type 0001:09 (IM:InvokeCommandResponse)
I (465353) chip[IN]: (S) Sending msg 64842726 on secure session with LSID: 21328
I (465373) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 64 **********
I (465373) chip[ZCL]: Event: move from 64
I (465383) chip[ZCL]:  to 219 
I (465383) chip[ZCL]: (diff +1)
I (465383) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 219 **********
I (465393) app_driver: [app_driver_attribute_update]
I (465403) app_driver: [M endpoint_id] 1
I (465403) app_driver: [M attribute_id] 0
I (465423) app_driver: [M cluster_id] 8
I (465423) app_driver: [app_driver_light_set_brightness] 86
I (465433) led_driver_ws2812: led set r:219, g:216, b:214
I (465443) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x0000000F is 0 **********
I (465443) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000005's Attribute 0x00000003 is 0 **********
I (465453) app_driver: [app_driver_attribute_update]
I (465463) app_driver: [M endpoint_id] 1
I (465463) app_driver: [M attribute_id] 3
I (465473) app_driver: [M cluster_id] 5
I (465473) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000001 is 0 **********
I (465483) app_driver: [app_driver_attribute_update]
I (465493) app_driver: [M endpoint_id] 1
I (465493) app_driver: [M attribute_id] 1
I (465503) app_driver: [M cluster_id] 8
I (465513) chip[EM]: >>> [E:57309r S:21328 M:240716391 (Ack:64842726)] (S) Msg RX from 1:000000000001B669 [193C] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (465523) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000008's Attribute 0x00000000 is 219 **********
I (465533) chip[EM]: <<< [E:63225i S:21328 M:64842727] (S) Msg TX to 1:000000000001B669 [193C] --- Type 0001:05 (IM:ReportData)
I (465553) chip[IN]: (S) Sending msg 64842727 on secure session with LSID: 21328
I (465553) chip[DMG]: Refresh Subscribe Sync Timer with min 1 seconds and max 10 seconds

**my  app_main.c**  
app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */
    nvs_flash_init();

    /* Initialize driver */
    app_driver_handle_t light_handle = app_driver_light_init();
    app_driver_handle_t button_handle = app_driver_button_init();
    app_reset_button_register(button_handle);

    /* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
    node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

#if 1
    extended_color_light::config_t light_config;
    light_config.on_off.on_off = DEFAULT_POWER;
    light_config.on_off.lighting.start_up_on_off = nullptr;
    light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
    light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
    // use EMBER_ZCL_COLOR_MODE_CURRENT_X_AND_CURRENT_Y cannot help
    light_config.color_control.color_mode = **EMBER_ZCL_COLOR_MODE_CURRENT_X_AND_CURRENT_Y**;
    light_config.color_control.enhanced_color_mode = **EMBER_ZCL_COLOR_MODE_CURRENT_X_AND_CURRENT_Y**;//;EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
    light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;
    light_config.color_control.xy.current_x = 0x616b;
    light_config.color_control.xy.current_y = 0;//0x616b;
    endpoint_t *endpoint = extended_color_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);
#endif

    /* These node and endpoint handles can be used to create/add other endpoints and clusters. */
    if (!node || endpoint) {
        ESP_LOGE(TAG, "Matter node creation failed");
    }

   light_endpoint_id = endpoint::get_id(endpoint);
    ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }

    /* Starting driver with default values */
    app_driver_light_set_defaults(light_endpoint_id);

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#endif
}

shubhamdp, 2023-06-26:
@zsr391912829 Can you please share the following details:
- esp-idf and esp-matter commit ids
- Devkit: (eg: ESP32, ESP32C3)
- Logs from the device (Please attach the log file, do not paste it)
- And any patches on top of current example to reproduce the issue.

Also, can you check if this works with host based chip-tool?

dhrishi, 2023-07-11:
@zsr391912829 Please share the requested details. Else, close the issue if resolved

dhrishi, 2023-07-27:
Closing due to no response

closed 2023-07-27 by dhrishi