espressif/esp-matter #484: Esp-Matter light example on ESP32-H2 cannot pair with HomePod Mini

Duncankn, 2023-06-29:
I have flashed the light example to my ESP32-H2-DevKitM-1 and tried to add it to HomeKit with my HomePod Mini.
The HomePod can detect the matter accessory.
However, I cannot add it to the HomeKit as the app always requests the code but cannot find any code when I monitor the running firmware. 
is there a link for the QR code generation during the firmware is running?

chshu, 2023-06-29:
The default QR code for the example is under this link:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning

Duncankn, 2023-06-29:
> The default QR code for the example is under this link: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#commissioning

Thanks for the prompt reply!
I have scanned this QR code but it's failed, so I drilled into the output.
are there any things wrong?

`ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x400031b6
0x400031b6: software_reset_cpu in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xc10
load:0x4083efd0,len:0x2b7c
load:0x408460e0,len:0x1838
entry 0x4083cfd0
I (22) boot: ESP-IDF v5.1-rc1-dirty 2nd stage bootloader
I (23) boot: compile time Jun 20 2023 15:04:45
I (24) boot: chip revision: v0.1
I (25) boot.esp32h2: SPI Speed      : 64MHz
I (30) boot.esp32h2: SPI Mode       : DIO
I (35) boot.esp32h2: SPI Flash Size : 4MB
I (40) boot: Enabling RNG early entropy source...
W (45) bootloader_random: bootloader_random_enable() has not been implemented yet
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (72) boot:  1 nvs              WiFi data        01 02 00010000 00006000
I (79) boot:  2 nvs_keys         NVS keys         01 04 00016000 00001000
I (86) boot:  3 otadata          OTA data         01 00 00017000 00002000
I (94) boot:  4 phy_init         RF data          01 01 00019000 00001000
I (101) boot:  5 ota_0            OTA app          00 10 00020000 001c0000
I (109) boot:  6 ota_1            OTA app          00 11 001e0000 001c0000
I (116) boot:  7 ot_storage       Unknown data     01 81 003a0000 00002000
I (124) boot: End of partition table
I (129) esp_image: segment 0: paddr=00020020 vaddr=42120020 size=383e8h (230376) map
I (236) esp_image: segment 1: paddr=00058410 vaddr=40800000 size=07c08h ( 31752) load
I (253) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=11d45ch (1168476) map
I (755) esp_image: segment 3: paddr=0017d484 vaddr=40807c08 size=08254h ( 33364) load
I (773) esp_image: segment 4: paddr=001856e0 vaddr=4080fe60 size=01ebch (  7868) load
I (788) boot: Loaded app from partition at offset 0x20000
I (790) boot: Disabling RNG early entropy source...
W (791) bootloader_random: bootloader_random_disable() has not been implemented yet
I (809) cpu_start: Unicore app
I (810) cpu_start: Pro cpu up.
W (820) clk: esp_perip_clk_init() has not been implemented yet
I (826) cpu_start: Pro cpu start user code
I (827) cpu_start: cpu freq: 96000000 Hz
I (827) cpu_start: Application information:
I (829) cpu_start: Project name:     light
I (834) cpu_start: App version:      v1.0
I (839) cpu_start: Compile time:     Jun 20 2023 15:04:38
I (845) cpu_start: ELF file SHA256:  c2a96b0ac0876ddf...
I (851) cpu_start: ESP-IDF:          v5.1-rc1-dirty
I (856) cpu_start: Min chip rev:     v0.0
I (861) cpu_start: Max chip rev:     v0.99 
I (866) cpu_start: Chip rev:         v0.1
I (870) heap_init: Initializing. RAM available for dynamic allocation:
I (878) heap_init: At 40826980 len 00026A00 (154 KiB): D/IRAM
I (884) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (892) spi_flash: detected chip: generic
I (896) spi_flash: flash io: dio
W (900) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
I (911) coexist: coex firmware version: ebddf30
I (916) app_start: Starting scheduler on CPU0
I (920) main_task: Started on CPU0
I (920) main_task: Calling app_main()
I (930) led_driver_ws2812: Initializing light driver
[... 63 lines cut ...]
I (1940) led_driver_ws2812: led set r:0, g:0, b:0
I (1950) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000300's Attribute 0x00004010 is null **********
I (1960) esp_matter_core: Dynamic endpoint 1 added
I (1970) led_driver_ws2812: led set r:0, g:0, b:0
I (1970) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (1980) chip[DL]: Device already advertising, stop active advertisement and restart
I (1980) NimBLE: GAP procedure initiated: stop advertising.

I (1980) led_driver_ws2812: led set r:0, g:0, b:0
I (2000) NimBLE: GAP procedure initiated: advertise; 
I (2000) NimBLE: disc_mode=2
I (2000) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (2020) NimBLE: 

I (2020) app_main: Commissioning window opened
I (2020) led_driver_ws2812: led set r:63, g:51, b:40
> I (2540) main_task: Returned from app_main()
I (31710) chip[DL]: Configuring CHIPoBLE advertising (interval 500 ms, connectable)
I (31710) chip[DL]: Device already advertising, stop active advertisement and restart
I (31720) NimBLE: GAP procedure initiated: stop advertising.

I (31720) NimBLE: GAP procedure initiated: advertise; 
I (31740) NimBLE: disc_mode=2
I (31740) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
I (31750) NimBLE: `

chshu, 2023-06-29:
@Duncankn The log looks good, the commissioner should establish BLE connection with the device after scanning QR code, but nothing happened. Make sure there is only one device powered on waiting for commissioning.

Duncankn, 2023-06-30:
Thanks! sometimes, I have different log with BLE GAP connection terminated(con 1 reason 0x208)

I (48710) chip[DL]: CHIPoBLE advertising started
I (48930) chip[DL]: BLE GAP connection established (con 1)
I (48930) chip[DL]: CHIPoBLE advertising stopped
I (49990) chip[DL]: BLE GAP connection terminated (con 1 reason 0x208)
I (50000) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (50000) NimBLE: GAP procedure initiated: advertise; 
I (50010) NimBLE: disc_mode=2
I (50010) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40

chshu, 2023-06-30:
This issue happens on some old iOS version, the latest iOS 16.5 should work, or you need to install some particular profile:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#profile-installation

Duncankn, 2023-07-11:
I successfully paired the dev board to the Homepod mini after following the guides.
Thanks a lot!

closed 2023-07-11 by Duncankn