espressif/esp-matter #485: example zigbee_bridge

Oscarren22, 2023-06-29:
I am testing the Matter Zigbee bridge example. The problem is, first of all, when configuring the Zigbee bulb on ESP32-H2, this example does not exist for esp-idf 5.1, but it does exist for the esp-idf 5.0 version. And here I encountered the second problem, which is why I cannot continue. This example does not support ESP-C6, and unfortunately, I do not have an ESP-H2. How can I configure the ESP-C6 to connect it to the Matter bridge via Zigbee

chshu, 2023-06-30:
The ESP32-C6 can also be used as a zigbee RCP, following the steps here:
https://github.com/espressif/esp-matter/blob/0cb1c14/examples/zigbee_bridge/README.md#12-build-and-flash-the-rcp-esp32-h2

Just set-target to esp32c6.

dhrishi, 2023-07-11:
@Oscarren22 Please give it a try and close the issue if it works

dhrishi, 2023-07-17:
Closing this now. Feel free to re-open in case of any updates

closed 2023-07-17 by dhrishi