espressif/esp-matter #488: Matter - OpenThread CLI

tommyjlong, 2023-06-30:
I have successfully commissioned the ESP32-H2 onto a Thread network using Matter.
I built the "Light" application according to [this link](https://docs.espressif.com/projects/esp-matter/en/latest/esp32h2/developing.html).
I built using target `idf.py --preview set-target esp32h2` and using default configs.
Gaining access to the ESP32's shell using `idf.py -p <PORT> monitor`,  I get:
```
> help

matter 
  Matter utilities

help 
  Print the list of registered commands

Done

> matter help

  esp             Usage: matter esp <sub_command>
  base64          Base64 encode / decode utilities
  exit            Exit the shell application
  help            List out all top level commands
  version         Output the software version
  ble             BLE transport commands
  config          Manage device configuration. Usage to dump value: config [param_name] and to set some values (discriminator): config [param_name] [param_value].
  device          Device management commands
  onboardingcodes Dump device onboarding codes. Usage: onboardingcodes none|softap|ble|onnetwork [qrcode|qrcodeurl|manualpairingcode]
  dns             Dns client commands
  ota             OTA commands
Done
```
After commissioning the device, I later found in the menuconfig ->Component config->OpenThread
```
[*] Enable Openthread Command-Line Interface
```
I enabled it as shown, saved it and rebuilt the project image.

In the `build/config/sdkconfig.h` I find:  `#define CONFIG_OPENTHREAD_CLI 1`
However after reflashing the ESP32-H2, I can not find a way to get to the OpenThread CLI.  Does the OpenThread CLI work?

chshu, 2023-07-04:
@tommyjlong OpenThread CLI is not available in Matter example, the Matter application takes over the Thread network configuration.

tommyjlong, 2023-07-06:
OK, I will close this.

closed 2023-07-06 by tommyjlong