espressif/esp-matter #489: how can a signature be integrated into an OTA image?

albrownwood, 2023-07-01:
The accessory will have a public key set in factory.
Each new image received from OTA should include a signature verifiable with that public key.
I couldn't find documentation about this, is it possible to get more info or guidance on how to do that?

Thanks

shubhamdp, 2023-07-04:
Are you taking in context with [secure boot v2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html)? Or you want to do this in your application? Also, which public key is this?

albrownwood, 2023-07-06:
in fact I am talking about secure boot v1 (since I am using an older esp32 hardware) and the possibility to combine it with Matter OTA images as described here https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/ota.md
The Matter OTA document above describes image encryption which is not what I am interested in, only image verification as provided by secure boot v1 is needed in my case.

shubhamdp, 2023-07-13:
@albrownwood Sorry for the late turnaround, I was out for a few days.

If you are using the secure boot then during boot up, application's signature verification will be taken care by the bootloader. Same verification happens during the OTA upgrade process.

To generate and integrate signature, you can use `espsecure.py` utility and steps mentioned in the [secure boot v1 guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#remote-signing-of-images).

[Signed App Verification Without Hardware Secure Boot](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v1.html#signed-app-verification-without-hardware-secure-boot) sections has more details about the working of secure boot. 

To use this with Matter OTA, you just need to build an application, sign the image, and prepend the Matter OTA header to the signed image and use this for OTA. Matter SDK will strip out the Matter OTA header and will pass on the rest of the image to app_update component which will verify the image signature if secure boot is enabled.

albrownwood, 2023-07-16:
@shubhamdp it all makes sense, thanks for the clear explanations! 
Closing this.

closed 2023-07-16 by albrownwood