espressif/esp-matter #490: ota_data_initial.bin

SYESHASWINI, 2023-07-03:
I'm just trying to understand what is the ota_data_initial.bin file that is in the build directory. What does this file contains.? Could any please provide the information regarding this. Thanks in advance.

SYESHASWINI, 2023-07-04:
@shubhamdp as no ota data is present in ota_data_initial.bin when the ota performed this bin file is not used or upgraded right? the only files which upgrade through ota is project.bin, bootloader.bin and partition.bin.
When devices are flashing using bin files could we eliminate this ota_data_initial.bin.

shubhamdp, 2023-07-04:
`ota_data_initial.bin`  is not upgraded with OTA, but updated using an API [esp_ota_set_boot_partition](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/OTAImageProcessorImpl.cpp#L344).

Right now, we only upgrade application, and not the partition table and bootloader.

> could we eliminate this ota_data_initial.bin

I'd suggest against it, to keep things simple and not to run into problems when using flash encryption. When flash encryption is disabled, erase flash will zero the stuff on the flash and not flashing the `ota_data_initial.bin` would work.

But, if you enable the flash encryption, `ota_data_initial.bin` has to be flashed after encrypting(This happens in `idf.py flash` flow). So, content after encrypting an empty `ota_data_initial.bin` will be different.

SYESHASWINI, 2023-07-04:
@shubhamdp Thanks for the information. When the ota is succesfully completed the bootloader loads the application from ota_1 instead of ota_0 so isn't the bootloader.bin is upgraded with ota process.

shubhamdp, 2023-07-04:
@SYESHASWINI Nope, OTA process only updates application.

SYESHASWINI, 2023-07-05:
Thanks @shubhamdp

closed 2023-07-05 by SYESHASWINI