espressif/esp-matter #492: Error when running ./install.sh on Apple Silicon M1

law-ko, 2023-07-04:
Hello,

I have issues when trying to set up the asp-matter environment on Apple M1 MacBook Air. 

Below is the error output when performing `./install.sh`

```
SelfTest/Signature/test_vectors/wycheproof
      running build_ext
      building 'Crypto.Hash._MD2' extension
      creating build/temp.macosx-10.9-universal2-cpython-39
      creating build/temp.macosx-10.9-universal2-cpython-39/src
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -Wno-error=unreachable-code -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/Users/irenelee/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/include -I/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/Headers -c src/MD2.c -o build/temp.macosx-10.9-universal2-cpython-39/src/MD2.o
      In file included from src/MD2.c:28:
      src/common.h:34:10: fatal error: 'stdio.h' file not found
         34 | #include <stdio.h>
            |          ^~~~~~~~~
      1 error generated.
      error: command '/Users/irenelee/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects
```

How to resolve this?

Thank you

ydanrubinstein, 2023-07-09:
gettng the same error (also on Mac M1)

law-ko, 2023-07-13:
I resolved it by removing all the repos (including ESP-IDF and ESP-Matter) and then re-install everything again

dhrishi, 2023-07-13:
@law-ko Thanks for the confirmation. Closing it now

closed 2023-07-13 by dhrishi