espressif/esp-matter #493: Unable to add matter devices to apple home

SYESHASWINI, 2023-07-05:
Hi, I recently updated my IOS version, from then my matter devices are unable to add to apple home. Could anyone help me with this. Versions I'm using are:
Home Pod: (MJ23HN/A(v 16.5)
iOS version - 16.5.1

Attaching the error log.
[apple_issue1.txt](https://github.com/espressif/esp-matter/files/11956950/apple_issue1.txt)

VaishaliAvhale, 2023-07-05:
I tried this and it is working on my side with the same Home Pod and iOS version.

VaishaliAvhale, 2023-07-06:
Have you made any changes in the light example?

SYESHASWINI, 2023-07-07:
@VaishaliAvhale  I haven't changed anything. I tried downloading the Matter Profile, still the device is not adding to the network. Its seems the BLE is getting terminated. How can I solve this.

VaishaliAvhale, 2023-07-10:
Can you try rebooting  HomePod,  ipad and factory reset the firmware ?

dhrishi, 2023-07-11:
@SYESHASWINI We are able to commission the device to Apple Home. I think it has to do something with your setup. I would recommend to file an issue to the feedback assistant (with the appropriate details and logs) by following the steps here: https://github.com/espressif/connectedhomeip/blob/master/docs/guides/darwin.md#providing-feedback-to-apple

closed 2023-07-11 by dhrishi