espressif/esp-matter #494: TC ACL 1.1 fails

AtiosAG, 2023-07-05:
I am testing our matter device using the Raspberry Pi TH suite, running test Test_TC_ACL_1_1. It is a bridge device (aggregator endpoint) commissioned over Wi-Fi/BLE. Pairing with e.g. Apple Homekit works, so I am suprised the test fails. Any ideas why this happens?

[image]
[UI_Test_Run_2023_7_5_12_42_35.log](https://github.com/espressif/esp-matter/files/11956982/UI_Test_Run_2023_7_5_12_42_35.log)

AtiosAG, 2023-07-05:
I repeated the test using onnetwork commissioning over ethernet. Pairing works perfectly fine with Apple Homekit, but chip-tool times out. Our device and the TH RPI are on the same network, they can ping each other. However, our device only gets an IPv4 address, can this be a issue? I also noticed that the ethernet examples in esp-idf provide only IPv4 capability and not IPv6...
I also tried to pair with manual commands, e.g. chip-tool pairing onnetwork <node-id> <pairing code>, but it times out too. Any hints from your side? 

[UI_Test_Run_2023_7_5_13_45_40.log](https://github.com/espressif/esp-matter/files/11957782/UI_Test_Run_2023_7_5_13_45_40.log)

VaishaliAvhale, 2023-07-13:
Please update your PICS file. Below is the reason for the test steps failure
6th step: You are trying to read the optional attribute and the optional attribute isn't supported
8th step: Getting 0 entries of GeneratedCommandList Attribute is expected because accesscontrol cluster not generating and accepting server or client-side commands as a result it is  0.

VaishaliAvhale, 2023-07-14:
Please use the attached ACL PICS file for the esp-matter bridge device.
[Access Control cluster.zip](https://github.com/espressif/esp-matter/files/12046943/Access.Control.cluster.zip)

dhrishi, 2023-07-21:
@AtiosAG Is your issue resolved? Please close if it is

dhrishi, 2023-07-27:
@AtiosAG I will close this now. Please re-open if you are still seeing the issue

closed 2023-07-27 by dhrishi