espressif/esp-matter #495: ESP32-C6 Cannot be provisioned as thread device

avidNewb, 2023-07-06:
I tried to edit the ESP32-C6 Light example to use the H2 settings (no wifi and thread).  The device is discovered and go through all of the connecting steps for Smartthings but fails at "Reginstering device with your account".  If the issue is not esp-matter, please advise.

wqx6, 2023-07-10:
Did you disable minimal mDNS for C6? For Thread device, we must use platform mDNS. And could you send the log of ESP32-C6?

dhrishi, 2023-07-11:
@avidNewb Please provide the requested details for further debugging on our side

dhrishi, 2023-07-21:
https://docs.espressif.com/projects/esp-matter/en/latest/esp32c6/developing.html#flashing-the-firmware should help. Closing this now. Please re-open if you still see the issue.

KorvinSzanto, 2023-08-28:
I'm seeing this same behavior with my ESP32-c6, here's the bit of my log that shows openthread related stuff https://gist.github.com/KorvinSzanto/c445e0ef59444ddfe06753d39e0a0579

The log entries look like this:
```
I(64903) OPENTHREAD:[N] Mle-----------: Attach attempt 1, AnyPartition
I(73053) OPENTHREAD:[N] Mle-----------: Attach attempt 1 unsuccessful, will try again in 0.294 seconds
```

I'm using the light example with openthread settings from the light_switch example with changes from the "flashing-the-firmware" link @dhrishi provided. I get the same behavior on both of my thread networks, one is made up of google wifi pros (I've set up my google developer stuff and have successfully connected the device with matter over wifi) and the other is a home assistant skyconnect. I get the same behavior whether the ESP32C6 is next to a router or the skyconnect or not.

dhrishi, 2023-08-28:
@KorvinSzanto Did you make these changes as mentioned in the documentation -

```
ESP32-C6 supports both the Wi-Fi and IEEE 802.15.4 radio, so you can run Wi-Fi or Thread matter example on it.

To enable Thread, you should change the menuconfig options to CONFIG_OPENTHREAD_ENABLED=y, CONFIG_ENABLE_WIFI_STATION=n, and CONFIG_USE_MINIMAL_MDNS=n.

To enable Wi-Fi. you should change the menuconfig options to CONFIG_OPENTHREAD_ENABLED=n, CONFIG_ENABLE_WIFI_STATION=y, and CONFIG_USE_MINIMAL_MDNS=y.
```

If yes, please share your sdkconfig and upload the device side console logs.

KorvinSzanto, 2023-08-28:
Yes I applied those changes as mentioned in my previous comment. Here's my sdkconfig: https://gist.github.com/KorvinSzanto/b3bb4c77ed53770fbf4e235bc82f8bdf

And my device side logs from my previous comment:
https://gist.github.com/KorvinSzanto/c445e0ef59444ddfe06753d39e0a0579

KorvinSzanto, 2023-09-01:
@dhrishi Can this issue be reopened?

wqx6, 2023-09-08:
Could you please check whether your ESP-IDF includes this commit 26c0a23598520d0233f8d383b20667183cebd0ba? (the recommended IDF v5.1.1 should include that commit) If yes, then the `.storage_partition_name` in `app_priv.h` should be changed to `nvs`or other `nvs` type partition. Otherwise it should be `ot_storage`(partition with type 0x3a).

dhrishi, 2023-09-21:
@KorvinSzanto Please close this issue if it works

dhrishi, 2023-09-30:
Closing this now

closed 2023-09-30 by dhrishi

andersve, 2024-02-04:
I have the exact same problem as described above. I am also seeing a lot of unsuccessful attempts as KorvinSzanto described, more then 10, all unsuccessful. It looks like the device managed to join the Thread network however, since after the failed commissioning process there is a continuous chatter like this:

`I(697818) OPENTHREAD:[I] MeshForwarder-: Received IPv6 UDP msg, len:105, chksum:04e3, ecn:no, from:b20b108ce768b616, sec:no, prio:net, rss:-87.0
I(697828) OPENTHREAD:[I] MeshForwarder-:     src:[fe80:0:0:0:b00b:108c:e768:b616]:19788
I(697828) OPENTHREAD:[I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788
I(697838) OPENTHREAD:[I] Mle-----------: Receive Advertisement (fe80:0:0:0:b00b:108c:e768:b616,0x5800)`

I have checked for the esp-idf commit mentioned above, it is present and `.storage_partion_name` is already set to nvs.

My matter network is based on AppleTVs and HomePods, I have a bunch of Thread and Wifi Matter devices already present and working well.

I could mention that the exact same project (which is the extended color light example with minor modifications) works well when running in WiFi mode and is able to be commissioned properly.

wqx6, 2024-02-06:
@andersve Could you please provide the IDF version you are using? There is a BLE-15.4 co-exist issue for ESP32-C6 on the IDF commit you are using, and it should be fix in tag v5.1.2. Could you switch the IDF version to v5.1.2 and try again?

andersve, 2024-02-08:
Thank you @wqx6 , that did indeed solve my problem!