espressif/esp-matter #496: ESP32-C6 Light example doesn't control light

avidNewb, 2023-07-06:
The ESP32-C6 has an onboard WS2812B.  The light example is configured to use pin 8 which is the data pin for the WS2812B but it is using ledc commands instead of RMT.  The result is that the WS2812B turns on to full white but is not controllable.

wqx6, 2023-07-10:
The [led_type](https://github.com/espressif/esp-matter/blob/main/device_hal/device/esp32c6_devkit_c/esp_matter_device.cmake#L7) for esp32c6_devkit_c is WS2812. Can you check it and test again?

sayondeep, 2023-07-13:
I am able to commission and control the onboard led on esp32-c6. Switch on/off , colour change and brightness control everything works fine.

dhrishi, 2023-07-13:
Closing this as the C6 devkitC is controllable

closed 2023-07-13 by dhrishi