espressif/esp-matter #497: temperature target not taken into account by thermostat

albrownwood, 2023-07-06:
HEllo,
I am currently testing a thermostat device in HEAT mode, the target temperature is initially 2110 and I set it on the phone's UI to 2670.
As per the log below, the phone correctly receives the new temperature (see references to "Attribute 0x00000012" in the log) and I see the new target temperature in the UI, then a few seconds later the temperature set in UI reverts to the initial value 2110.

It seems like sometimes the new target temperature is correctly registered, sometimes it reverts depending on the difference between the initial and desired target value entered in the phone's UI.

Noite: the "app_attribute_update_cb" returns ESP_OK in all cases.

What am I missing?

`I (1455027) chip[EM]: <<< [E:27600r S:51412 M:12257215 (Ack:41318408)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1455037) chip[IN]: (S) Sending msg 12257215 on secure session with LSID: 51412
I (1455057) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2110 **********
I (1455067) chip[EM]: <<< [E:27601r S:51412 M:12257216 (Ack:41318410)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1455077) chip[IN]: (S) Sending msg 12257216 on secure session with LSID: 51412
I (1455097) chip[EM]: >>> [E:27600r S:51412 M:41318411 (Ack:12257215)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1455107) chip[EM]: >>> [E:27602r S:51412 M:41318412] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:06 (IM:WriteRequest)
I (1455117) esp_matter_attribute: ********** W : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2670 **********
I (1455127) app_main: PRE_UPDATE
I (1455127) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000FFFC is 51 **********
I (1455147) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000019 is 25 **********
I (1455157) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000005 is 1600 **********
I (1455167) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000006 is 3200 **********
I (1455177) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000017 is 1600 **********
I (1455197) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000018 is 3200 **********
I (1455207) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000003 is 700 **********
I (1455217) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000004 is 3000 **********
I (1455237) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000015 is 700 **********
I (1455247) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000016 is 3000 **********
I (1455257) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000011 is 2600 **********
I (1455267) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2110 **********
I (1455287) chip[EM]: <<< [E:27602r S:51412 M:12257217 (Ack:41318412)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:07 (IM:WriteResponse)
I (1455297) chip[IN]: (S) Sending msg 12257217 on secure session with LSID: 51412
I (1455307) chip[EM]: >>> [E:27601r S:51412 M:41318413 (Ack:12257216)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1455327) chip[EM]: >>> [E:27602r S:51412 M:41318414 (Ack:12257217)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1455697) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (1455707) chip[EM]: <<< [E:28998i S:51412 M:12257218] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1455707) chip[IN]: (S) Sending msg 12257218 on secure session with LSID: 51412
I (1455717) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (1455737) chip[EM]: >>> [E:28998i S:51412 M:41318415 (Ack:12257218)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:01 (IM:StatusResponse)
I (1455747) chip[IM]: Received status response, status is 0x00
I (1455757) chip[EM]: <<< [E:28998i S:51412 M:12257219 (Ack:41318415)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1455767) chip[IN]: (S) Sending msg 12257219 on secure session with LSID: 51412
I (1457727) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (1457727) chip[EM]: <<< [E:28999i S:51412 M:12257220] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1457737) chip[IN]: (S) Sending msg 12257220 on secure session with LSID: 51412
I (1457747) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (1457767) chip[EM]: >>> [E:28999i S:51412 M:41318416 (Ack:12257220)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:01 (IM:StatusResponse)
I (1457787) chip[IM]: Received status response, status is 0x00
I (1457787) chip[EM]: <<< [E:28999i S:51412 M:12257221 (Ack:41318416)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1457807) chip[IN]: (S) Sending msg 12257221 on secure session with LSID: 51412
I (1459757) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (1459757) chip[EM]: <<< [E:29000i S:51412 M:12257222] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1459767) chip[IN]: (S) Sending msg 12257222 on secure session with LSID: 51412
I (1459777) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (1459797) chip[EM]: >>> [E:29000i S:51412 M:41318417 (Ack:12257222)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:01 (IM:StatusResponse)
I (1459807) chip[IM]: Received status response, status is 0x00
I (1459807) chip[EM]: <<< [E:29000i S:51412 M:12257223 (Ack:41318417)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1459827) chip[IN]: (S) Sending msg 12257223 on secure session with LSID: 51412
I (1461777) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (1461787) chip[EM]: <<< [E:29001i S:51412 M:12257224] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1461787) chip[IN]: (S) Sending msg 12257224 on secure session with LSID: 51412
I (1461807) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (1461817) chip[EM]: >>> [E:29001i S:51412 M:41318418 (Ack:12257224)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:01 (IM:StatusResponse)
I (1461837) chip[IM]: Received status response, status is 0x00
I (1461837) chip[EM]: <<< [E:29001i S:51412 M:12257225 (Ack:41318418)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1461847) chip[IN]: (S) Sending msg 12257225 on secure session with LSID: 51412
I (1463807) chip[DMG]: Refresh subscribe timer sync after 2 seconds
I (1463817) chip[EM]: <<< [E:29002i S:51412 M:12257226] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
I (1463817) chip[IN]: (S) Sending msg 12257226 on secure session with LSID: 51412
I (1463827) chip[DMG]: Refresh Subscribe Sync Timer with min 0 seconds and max 2 seconds
I (1463847) chip[EM]: >>> [E:29002i S:51412 M:41318419 (Ack:12257226)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:01 (IM:StatusResponse)
I (1463857) chip[IM]: Received status response, status is 0x00
I (1463857) chip[EM]: <<< [E:29002i S:51412 M:12257227 (Ack:41318419)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1463877) chip[IN]: (S) Sending msg 12257227 on secure session with LSID: 51412
I (1465327) chip[EM]: >>> [E:27603r S:51412 M:41318420] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:02 (IM:ReadRequest)
I (1465337) chip[EM]: >>> [E:27604r S:51412 M:41318421] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:02 (IM:ReadRequest)
I (1465347) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x0000001C is 4 **********
I (1465367) chip[EM]: <<< [E:27603r S:51412 M:12257228 (Ack:41318420)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)
[... cut ...]
I (1465507) chip[EM]: >>> [E:27605r S:51412 M:41318425 (Ack:12257230)] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0000:10 (SecureChannel:StandaloneAck)
I (1465527) chip[EM]: >>> [E:27606r S:51412 M:41318426] (S) Msg RX from 1:311D76169C0F471B [00CD] --- Type 0001:02 (IM:ReadRequest)
I (1465537) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000201's Attribute 0x00000012 is 2110 **********
I (1465547) chip[EM]: <<< [E:27606r S:51412 M:12257231 (Ack:41318426)] (S) Msg TX to 1:311D76169C0F471B [00CD] --- Type 0001:05 (IM:ReportData)`

albrownwood, 2023-07-08:
more details:
it looks like in COOL mode I cannot set the desired temperature below a certain value (24C), however as long as I stay above that value I can change the desired temperature without issue.
in HEAT mode I cannot set the temperature above a certain point (21C) , as long as I stay below that point I can change the desired temperature without issue.
Not sure if it matters but the local temperature is set to 24C on the thermostat for that test.
There seem to be a constraint that is being applied here but I can't tell where.
The endpoint is created with the default configs as follow:
```
node::config_t node_config;
    node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

    thermostat::config_t thermostat_config;
    endpoint_t *endpoint = thermostat::create(node, &thermostat_config, ENDPOINT_FLAG_NONE, NULL);
    cluster_t *thermostat_cluster = cluster::get(endpoint, Thermostat::Id);
   
   //heat
   cluster::thermostat::feature::heating::config_t heat_config;
   cluster::thermostat::feature::heating::add(thermostat_cluster, &heat_config);

    //cool 
   cluster::thermostat::feature::cooling::config_t cool_config;
   cluster::thermostat::feature::cooling::add(thermostat_cluster, &cool_config);    

    //auto
   cluster::thermostat::feature::auto_mode::config_t auto_config;
   cluster::thermostat::feature::auto_mode::add(thermostat_cluster, &auto_config); 

   //setback
   cluster::thermostat::feature::setback::config_t setback_config;
   cluster::thermostat::feature::setback::add(thermostat_cluster, &setback_config);
```

albrownwood, 2023-07-11:
[set_th.txt](https://github.com/espressif/esp-matter/files/12009439/set_th.txt)

at t=45598 attempt to set attribute 0x11 to value 2300, current value is 3000
callback never fires with POST_UPDATE event
at t= 55348  attribute is still at 3000 and being reverted to 3000 in phone UI

albrownwood, 2023-07-17:
Hi @VaishaliAvhale, is there any progress on this issue? is the issue in the esp-matter stack or in the CHIP stack?

shubhamdp, 2023-07-20:
@albrownwood We are looking into the problem, will drop an update here, soon.

VaishaliAvhale, 2023-10-16:
@albrownwood It is Google's side issue. The fix will be included in the next Nest Hub software release (f14). It will happen in the mid of November.

VaishaliAvhale, 2024-02-22:
Nest Hub Software Version F14 is released and temperature target setting working as expected please check!

dhrishi, 2024-04-24:
Closing this. Please give it a try and re-open if it does not work as expected

closed 2024-04-24 by dhrishi