espressif/esp-matter #498: ESP32 S2 not supported

stk84, 2023-07-06:
Hi!
I couldn't find any documentation on this so unfortunately i have to ask here.

Why is the ESP32 S2 excluded for all of the examples (see CMakeLists.txt check for "IDF_TARGET") ? 

Is it due to a missing BLE Chip? 
IP Commisioning should still be an alternative, right?

jonsmirl, 2023-07-07:
S2 has some pretty severe RAM limitations, I'm not sure Matter will fit and be useful. The ESP32-C3 is pretty similar and Matter works well on it.  It's even cheaper than the S2.

dhrishi, 2023-07-10:
@stk84 

>IP Commisioning should still be an alternative, right?

Yes, IP commissioning should still work. You can test it out. Let us know in case you see any issues

dhrishi, 2023-07-20:
S2 has not been added as the SoCs with BLE are added as the default ones. Although, you can add it for your application and it should just work fine. Please re-open this issue if it doesn't.

closed 2023-07-20 by dhrishi