espressif/esp-matter #500: CMake error

Okywan, 2023-07-07:
Hello,

By using the CMakeLists.txt you're using on the examples code doesn't compile:

CMake Error at CMakeLists.txt:37 (include):
  include could not find requested file:

   $ENV{IDF_PATH}/tools/esp-matter/device_hal/device/esp_matter_device.cmake

contents of CMakeLists.txt:

`# For more information about build system see
# https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

if(NOT DEFINED ENV{ESP_MATTER_PATH})
    message(FATAL_ERROR "Please set ESP_MATTER_PATH to the path of esp-matter repo")
endif(NOT DEFINED ENV{ESP_MATTER_PATH})

if(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})
    if("${IDF_TARGET}" STREQUAL "esp32" OR "${IDF_TARGET}" STREQUAL "")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32c3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c3_devkit_m)
    elseif("${IDF_TARGET}" STREQUAL "esp32c2")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c2_devkit_m)
    elseif("${IDF_TARGET}" STREQUAL "esp32h2")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32h2_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32s3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32s3_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32c6")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c6_devkit_c)
    else()
        message(FATAL_ERROR "Unsupported IDF_TARGET")
    endif()
endif(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})

set(PROJECT_VER "v1.0")
set(PROJECT_VER_NUMBER 1) 

set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)

# This should be done before using the IDF_TARGET variable.
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include($ENV{ESP_MATTER_DEVICE_PATH}/esp_matter_device.cmake) #<<<------ ERROR IS HERE; LINE 37
set(EXTRA_COMPONENT_DIRS 
    "./components"
    "${ESP_MATTER_PATH}/examples/common"
    "${MATTER_SDK_PATH}/config/esp32/components"
    "${ESP_MATTER_PATH}/components"
    "${ESP_MATTER_PATH}/device_hal/device"
    ${extra_components_dirs_append}
)

#idf_build_component(components/IRDriver/)
#idf_build_component(components/WifiDriver/)
project(ac-controller-matter)

idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++14;-Os;-DCHIP_HAVE_CONFIG_H" APPEND)
idf_build_set_property(C_COMPILE_OPTIONS "-Os" APPEND)
# For RISCV chips, project_include.cmake sets -Wno-format, but does not clear various
# flags that depend on -Wformat
idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security" APPEND)`

dhrishi, 2023-07-11:
>By using the CMakeLists.txt you're using on the examples code doesn't compile:

@Okywan Which example are you trying to build? Please provide details about branch/commit id and the set of steps you followed

Okywan, 2023-07-11:
@dhrishi, I'm trying to compile my own code, not an example. I downloaded and installed the esp-matter repo into my tools folder. This is the commit id:

esp-matter % git log -n 1
commit 7be04110831b25ad803655b16f0a04f371438d55 (HEAD -> main, origin/main, origin/HEAD)
Merge: f31db37 1b3cf7e
Author: Shu Chen <chenshu@espressif.com>
Date:   Thu Jun 22 20:20:04 2023 +0800

    Merge branch 'bugfix/blemesh_bridge_DAC_compile' into 'main'
    
    blemesh_bridge: compile failed when enable secure cert DAC provider
    
    See merge request app-frameworks/esp-matter!388

This line in the CMakeLists crashes the compiler because there is no such file:
include($ENV{ESP_MATTER_DEVICE_PATH}/esp_matter_device.cmake)

dhrishi, 2023-07-12:
@Okywan Do you have anything similar to [this](https://github.com/espressif/esp-matter/blob/main/examples/light/CMakeLists.txt#L9-L25) in your example top level CMakeLists.txt? This sets the `ESP_MATTER_DEVICE_PATH` and then the corresponding file (say for ESP32) is chosen from here: https://github.com/espressif/esp-matter/blob/main/device_hal/device/esp32_devkit_c/esp_matter_device.cmake

Okywan, 2023-07-14:
Yes, it's there. I'm attaching the file. Problem seems to be it's not choosing `device_hal/device/esp32_devkit_c/` but `
[CMakeLists.txt](https://github.com/espressif/esp-matter/files/12053560/CMakeLists.txt)
/device_hal/device/`

Also, now it complains with the first check for ESP_MATTER_PATH, but it is defined:

esp-idf % echo $ESP_MATTER_PATH
~/proyectos/esp32/esp-idf/tools/esp-matter

shubhamdp, 2023-08-02:
@Okywan You may have forgot to source the esp-matter/export.sh. Can you please run the `cd esp-matter &&  source export.sh`

Okywan, 2023-08-06:
No... it was sourced... printenv shows the esp matter path.

shubhamdp, 2023-08-07:
Is esp-matter clonned inside the `esp-idf/tools` repository? If yes, could be a problem. Can you move esp-matter outside esp-idf repository?

Okywan, 2023-08-07:
Yes, it is there but if that can cause issues why does VSCode extension install esp-matter command install it there?!

shubhamdp, 2023-08-11:
> why does VSCode extension install esp-matter command install it there?

VSCode Extension does not install esp-matter there. It asks user where to download it, please check the 4th point from the extension documentation. https://github.com/espressif/vscode-esp-idf-extension#about-commands

I have seem the similar issue earlier, no idea why. One reason could be, `esp-idf/tools` contains the build system bits. I'd suggest you move that out of esp-idf and give a try.

dhrishi, 2023-09-05:
Closing this as it is addressed

closed 2023-09-05 by dhrishi