espressif/esp-matter #501: add matter to espressif-ide

amin31, 2023-07-09:
how can I add matter to espressif-ide? I download Matter and tried to compile a code but the error "Please set ESP_MATTER_PATH to the path of esp-matter repo" was shown.

shripad621git, 2023-07-20:
To resolve the error  "Please set ESP_MATTER_PATH to the path of esp-matter repo", you need to export the path of esp-matter repository. 
Traverse to the esp-matter repository and run the command ```. ./export.sh```.
For more help regarding building and trying out examples please refer https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html.

sayondeep, 2023-08-31:
@amin31 
You can add ESP_MATTER_PATH into espressif-ide by going on the `window-->Preferences-->C/C++-->Build-->Environment` the Add the following Variables:
ESP_MATTER_PATH : path/to/esp-matter
ZAP_INSTALL_PATH : path/to/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/zap

Append the following lines in PATH variable by using ' : ' between each value 
/path/to/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/:/path/to/esp-matter/connectedhomeip/connectedhomeip/out/host

You can also add IDF_TARGET : esp32c3 (according to your device)

amin31, 2023-08-31:
tnx my problem got solved

closed 2023-08-31 by amin31

S-A-Hussaini, 2024-05-22:
I have done these workarounds. There is one dependency that project directory must have a .git folder else project build is failing. For testing purpose I added a dummy .git file and build process moved one step ahead (Is there anyway that we can remove .git dependency?).
I tried all fixes, still build is failing. Following is the error.

**-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
ESP Insights Project commit: HEAD-HASH-NOTFOUND
CMake Error at /home/oem/esp/esp-idf-v5.2.1/tools/cmake/component.cmake:281 (message):
  SRC_DIRS entry
  'null:/home/oem/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/attributes'
  does not exist.
Call Stack (most recent call first):
  /home/oem/esp/esp-idf-v5.2.1/tools/cmake/component.cmake:443 (__component_add_sources)
  /home/oem/esp/esp-matter/components/esp_matter/CMakeLists.txt:33 (idf_component_register)
-- Configuring incomplete, errors occurred!
See also "/home/oem/ec_workspaces/ec_matter_ws/light_test/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': No such file or directory**

I have cross verified the existence of attributes directory.

sayondeep, 2024-05-22:
@S-A-Hussaini Did you make any changes to the CMakeLists.txt of the example?

S-A-Hussaini, 2024-05-22:
Yes I have updated the files path accordingly.

S-A-Hussaini, 2024-05-22:
Thanks for the reference. I have followed all the steps as you mentioned. But still I am getting the same build error.
 SRC_DIRS entry
  'null:/home/oem/esp/esp-matter/connectedhomeip/connectedhomeip/zzz_generated/app-common/app-common/zap-generated/attributes'
  does not exist.

sayondeep, 2024-05-22:
@S-A-Hussaini 
Have you installed esp-matter?
Can you build the light example from the terminal ?

S-A-Hussaini, 2024-05-22:
YES I have installed ESP-MATTER SDK and successfully able to build the light example in Linux terminal.

sayondeep, 2024-05-22:
@S-A-Hussaini I suspect you may have missed a '/' at the end..somewhere in configuring the environment variables. Can you please try and check.

S-A-Hussaini, 2024-05-23:
@sayondeep Thanks a lot for guidance, issue is resolved.

athijegannathan, 2024-08-01:
how can I add matter to espressif-ide? I download Matter and tried to compile a code but the error "Please set ESP_MATTER_PATH to the path of esp-matter repo" was shown.

sayondeep, 2024-08-02:
> how can I add matter to espressif-ide? I download Matter and tried to compile a code but the error "Please set ESP_MATTER_PATH to the path of esp-matter repo" was shown.

https://github.com/espressif/esp-matter/issues/501#issuecomment-1700432298

V-S-Naidu, 2025-11-07:
Hi,

I am currently working on the Thread Border Router. For this purpose, I have used the basic Thread Border Router example from the Matter SDK release v1.4 and flashed it onto the ESP32-S3 chipset. After flashing, I am continuously receiving the attached log. Could you please suggest a possible solution for this?

Debug log : 

--- esp-idf-monitor 1.8.0 on /dev/ttyUSB0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x18c4
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc80
load:0x403cb700,len:0x2f0c
entry 0x403c8914
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037f7aa
--- 0x4037f7aa: esp_cpu_wait_for_intr at /home/vamsi/esp/esp-idf-v5.4.1/components/esp_hw_support/cpu.c:64
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x18c4
load:0x403c8700,len:0x4
load:0x403c8704,len:0xc80
load:0x403cb700,len:0x2f0c
entry 0x403c8914

V-S-Naidu, 2025-11-12:
@Jerry-ESP 

Thanks for the Response.

Could you please explain the difference between this mentioned Thread Border Router development board ? I currently have an ESP32-S3-M0N8R8 connected to an ESP32-H2-M8N4 via UART. For both setups, we need to flash the firmware separately, correct?  

If there is any difference, please let me know so that I can organize them accordingly.