espressif/esp-matter #503: Can not received Sigma1 msg from Alexa Dot

sidhoo2, 2023-07-11:
Hi,

Alexa APP info
  name：Echo Dot-4V1
  version：0009094444164
  Alexa ver：App version 2023.13 build version 2.2.517011.0

Alexa fails to add Matter devices, but our matter products can be added by Apple Home/Google Home, and there will be no prompt that the device is not certified. the device logs indicate that  no Sigma1 msg from Alexa has been received. 

[Alexa.log](https://github.com/espressif/esp-matter/files/12012965/Alexa.log)

dhrishi, 2023-07-11:
@sidhoo2 Can you please check if you can see the device advertising over mDNS once it connects to the AP. You can use the below command from a host machine present on the same Wi-Fi network as that of the device -
```
avahi-browse _matter._tcp -r
```
If you see your device in the response (which you should, ideally) then your observation would be correct  -

>no Sigma1 msg from Alexa has been received.

Also, this seems to be a different example. Can you please try with esp-matter light example just to confirm the working?

```
[15:47:03]I (32415) CHIP[DL]: NimBLE Config ecoflow ble adv data...
[15:47:03]I (32417) EF-I: [ef_advertise_data_config:357]sn:HW52ZDH4SF541658, cpu id:<348518cc61c4HW52ZDH4SF541658>
[15:47:03]I (32418) CHIP[DL]: NimBLE start advertising...
[15:47:03]I (32418) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
[15:47:03]E (32419) chip[DL]: ble_gap_adv_start() failed: Error CHIP:0x0000000B
[15:47:03]E (32420) chip[DL]: Start advertising failed: Error CHIP:0x0000000B
[15:47:03]E (32422) chip[DL]: Disabling CHIPoBLE service due to error: Error CHIP:0x0000000B
```
Do you use BLE while the commissioning is going on?

dhrishi, 2023-07-14:
@sidhoo2 

>So we reset the Router and turn on IPv6, it works !

So, you mean, when you enable IPv6 on the router then the commissioning with Alexa works fine?

dhrishi, 2023-07-26:
Closing this now as the commissioning was successful

closed 2023-07-26 by dhrishi

sidhoo2, 2023-07-29:
> you mean, when you enable IPv6 on the router then the commissioning with Alexa works fine?
yes, long press the RESET button，enable IPv6 , it work with Alexa