espressif/esp-matter #505: Failed to get wifi scan results while Matter started

eme-bennytang, 2023-07-12:
In my code I would like to trigger a wifi scan to see the RSSI of the specific AP.
Normally without matter, I can just call 
=============
esp_wifi_init()
esp_wifi_set_mode(WIFI_MODE_STA)
esp_wifi_start()
---------------------------
esp_wifi_scan_start
esp_wifi_scan_get_ap_num
esp_wifi_scan_get_ap_records
==============

But after starting matter, I realize the wifi layer is controlled by matter stack
(ConnectivityManagerImpl_WIFI & NetworkCommissioningDriver)
So I instead, just call the scan start function
I got WIFI_EVENT_SCAN_DONE on the event callback sometime later
Then I call the get results functions
But I found that the RSSI information in the ap records is always 0

How can I get this RSSI like what I would do normally?

IDF 5.0.2
Matter SDK: release v1.1 commit 548ef58b604ef1df208e4a040803a4218e213103

wqx6, 2023-07-13:
https://github.com/project-chip/connectedhomeip/pull/27161 this PR fixed the issue but it is not added to release v1.1 branch. You can try the latest main branch. The issue should be fixed in that branch.

dhrishi, 2023-07-27:
@eme-bennytang The above issue is merged into upstream and the submodule in esp-matter is updated to include that

closed 2023-07-27 by dhrishi