espressif/esp-matter #506: How to create leaf endpoints under another endpoint in ESP-matter SDK ?

aggaddam, 2023-07-12:
I want to create a leaf endpoint under another endpoint. Is there a way we can make this in ESP-Matter SDK?  

Here is the sample data model that I want to create 

[image]

jadhavrohit924, 2023-07-12:
Yes you can create an endpoint under another enpoint. You can use this API [set_parent_endpoint](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L274) which will update the partsList of the parent endpoint.

aggaddam, 2023-07-12:
@jadhavrohit924 
Is it the right way to configure the composite endpoints ?

`app_driver_handle_t light_handle = app_driver_light_init();

/* Create a Matter node and add the mandatory Root Node device type on endpoint 0 */
node::config_t node_config;
node_t *node = node::create(&node_config, app_attribute_update_cb, app_identification_cb);

extended_color_light::config_t light_config;
light_config.on_off.on_off = DEFAULT_POWER;
light_config.on_off.lighting.start_up_on_off = nullptr;
light_config.level_control.current_level = DEFAULT_BRIGHTNESS;
light_config.level_control.lighting.start_up_current_level = DEFAULT_BRIGHTNESS;
light_config.color_control.color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
light_config.color_control.enhanced_color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
light_config.color_control.color_temperature.startup_color_temperature_mireds = nullptr;
endpoint_t *endpoint1 = extended_color_light::create(node, &light_config, ENDPOINT_FLAG_NONE, light_handle);

/* These node and endpoint handles can be used to create/add other endpoints and clusters. */
if (!node || !endpoint1) {
    ESP_LOGE(TAG, "Matter node creation failed");
}

light_endpoint_id = endpoint::get_id(endpoint1);
ESP_LOGI(TAG, "Light created with endpoint_id %d", light_endpoint_id);

endpoint_t *endpoint2 = endpoint::create(node, ENDPOINT_FLAG_NONE, NULL);
if (!endpoint2){
    ESP_LOGE(TAG, "EP2 creation failed");
}

tempsense_endpoint_id = endpoint::get_id(endpoint2);

temperature_measurement::config_t temperature_measurement_config;
cluster_t *cluster = temperature_measurement::create(endpoint2, &temperature_measurement_config, CLUSTER_FLAG_SERVER);

if (ESP_OK != set_parent_endpoint(endpoint2, endpoint1)){
    ESP_LOGE(TAG, "Composite Endpoint failed");
}

/* Matter start */
err = esp_matter::start(app_event_cb);
if (err != ESP_OK) {
    ESP_LOGE(TAG, "Matter start failed: %d", err);
}`

jadhavrohit924, 2023-07-12:
> endpoint_t *endpoint2 = endpoint::create(node, ENDPOINT_FLAG_NONE, NULL);
if (!endpoint2){
    ESP_LOGE(TAG, "EP2 creation failed");
}

I would suggest don’t use this API directly. If you want to `temperature_measurement` cluster, we have `Temperature Sensor`  device type for that. Code should be:
```c++
temperature_sensor::config_t temp_config;
endpoint_t *endpoint2 = temperature_sensor::create(node, &temp_config, ENDPOINT_FLAG_NONE, NULL);
if (!endpoint2){
    ESP_LOGE(TAG, "EP2 creation failed");
}

```  

Rest of the things looks good.

jonsmirl, 2023-07-12:
You can make them into a composite device, but then Apple/Google won't understand them. If you make them two separate devices, it will work in Google/Apple.  It is perfectly legal to make the composite device. It is a shortcoming of Apple/Google that their apps don't recognize composite devices.

I initially built composite devices and then tore them out for this reason.

closed 2023-07-20 by aggaddam