espressif/esp-matter #508: switch_on_off like light_on_off

stefano055415, 2023-07-12:
is it possible to create an object of type switch that controls two gpio?

that has the same behavior as the light_switch example, but display as a generic switch?

jonsmirl, 2023-07-12:
Sure, use strip the light example down to just OnOff (get rid of color and level control). Then make two OnOff endpoints, one for each GPIO. The switch side does not need to change. Just bind the switch to one of the two endpoints.

stefano055415, 2023-07-13:
@jonsmirl 

Hi, maybe I explained it wrong.

I would like to create a switch similar to the homekit one

which does not read the inputs but switches an output on and off

[image]

jonsmirl, 2023-07-13:
That is the light_switch example in the example folder.

AshishESP, 2023-07-19:
Does this mean your hardware does not have any input buttons?
and you want to operate the switch only using App or voice(alexa)?

jonsmirl, 2023-07-19:
If you want to do this in an app, you can look at the Google sample app
https://github.com/google-home/sample-app-for-matter-android

Here is where it sends the on/off commands
https://github.com/google-home/sample-app-for-matter-
android/blob/main/app/src/main/java/com/google/homesampleapp/chip/ClustersHelper.kt#L481

shubhamdp, 2023-08-02:
@stefano055415 any updates on this?

dhrishi, 2023-08-10:
Closing now. Please re-open in case of anything

closed 2023-08-10 by dhrishi