espressif/esp-matter #509: Using CHIP to send custom data?

law-ko, 2023-07-13:
Hello,

Is it possible to use CHIP to send custom data? Like how it is being done in [Espressif's provisioning solution](https://github.com/espressif/esp-idf/tree/d2471b11e78fb0af612dfa045255ac7fe497bea8/examples/provisioning/wifi_prov_mgr#sending-custom-data).

Thank you

dhrishi, 2023-07-13:
@law-ko Yes, you can send the custom data through the Matter data model (cluster, attributes, commands). The only requirement is that, as it is not standard data, both the sides (commissioner as well as the commissionee) should understand the data. You can check [this](https://github.com/espressif/esp-matter/blob/main/components/esp_matter_rainmaker/esp_matter_rainmaker.cpp#L290) for reference.

law-ko, 2023-07-26:
@dhrishi We have followed the reference you provided and are unable to locate the clusters which we have created under endpoint 0x0.

Locating the attribute by Matter Console in ESP32:
`matter esp attribute get 0x0 0xAA11FC00 0x0`
```E (205509) esp_matter_attribute: Matter attribute not found
Error: 83886338
```

We are initializing the clusters before Matter start:
```
    /* CustomAPI start */
    err = esp_matter::customapi::init();
    err = esp_matter::customapi::start();

    /* Matter start */
    err = esp_matter::start(app_event_cb);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "Matter start failed: %d", err);
    }
```

The custom API namespace:
```

namespace esp_matter {

    namespace cluster {
        namespace customapi {
            static constexpr chip::EndpointId endpoint_id = 0x00000000; /* Same as root node endpoint. This will always be
                                                                        endpoint_id 0. */
            static constexpr chip::ClusterId Id = 0xAA11FC00; /* 0xAA11 == manufacturer code. 
                                                                 0xFC00 == start of manufacuturer specific cluster_id */

            namespace attribute {
                namespace status {
                    static constexpr chip::AttributeId Id = 0x0000;
                } /* status */
                namespace node_id {
                    static constexpr chip::AttributeId Id = 0x0001;
                } /* node_id */
                namespace challenge {
                    static constexpr chip::AttributeId Id = 0x0002;
                } /* challenge */
            } /* attribute */

            namespace command {
                namespace configuration {
                    static constexpr chip::CommandId Id = 0x0000;
                } /* configuration */
                namespace sign_data {
                    static constexpr chip::CommandId Id = 0x0001;
                } /* sign_data */
            } /* command */
        } /* customapi */
    } /* cluster */

    namespace customapi {

        static esp_err_t status_attribute_update(bool status)
        {
            uint16_t endpoint_id = cluster::customapi::endpoint_id;
            uint32_t cluster_id = cluster:: customapi::Id;
            uint32_t attribute_id = cluster:: customapi::attribute::status::Id;
            esp_matter_attr_val_t val = esp_matter_bool(status);
            return attribute::update(endpoint_id, cluster_id, attribute_id, &val);
        }

        static esp_err_t node_id_attribute_update(char *node_id)
        {
            if (!node_id)
[... 63 lines cut ...]
            /* Create custom challenge attribute */
            /* Update the value of the attribute */
            char challenge[ESP_MATTER_CUSTOMAPI_MAX_CHALLENGE_LEN] = {0};
            attribute::create(cluster, cluster:: customapi::attribute::challenge::Id, ATTRIBUTE_FLAG_NONE, esp_matter_char_str(challenge, sizeof(challenge)));

            /* Create custom configuration command */
            command::create(cluster, cluster:: customapi::command::configuration::Id, COMMAND_FLAG_ACCEPTED | COMMAND_FLAG_CUSTOM, command_callback);

            return ESP_OK;
        }

        esp_err_t init()
        {
            return custom_cluster_create();
        }

        esp_err_t start()
        {
            return status_attribute_update(true);
        }
    } /* customapi */
} /* esp_matter */
```

Thank you.

sayondeep, 2023-07-28:
@law-ko As per [specification](https://csa-iot.org/developer-resource/specifications-download-request/) ClusterID is composed of Manufacturer Extensible Identifier (MEI) which has a prefix and a suffix. For prefix (0x0001 - 0xFFF0) is reserved for Manufacturers. You can try using 0x0000 as prefix. For Manufacturer-Specific Clusters ClusterID suffix can range between 0xFC00 - 0xFFFE. For, example 0x0000FCAB should work.

dhrishi, 2023-07-29:
@law-ko This should ideally solve your problem. Please close the issue if this does

dhrishi, 2023-08-10:
Closing this now

closed 2023-08-10 by dhrishi