espressif/esp-matter #510: esp-aws-iot submodule includes

law-ko, 2023-07-13:
Hello,

We are trying to include `esp-aws-iot` as a submodule and seems like the `backoffAlgorithm` are not included during the CMake building process. 

`idf.py build` result:
```
../main/networking/mqtt/core_mqtt_agent_manager.c:50:10: fatal error: backoff_algorithm.h: No such file or directory
   50 | #include "backoff_algorithm.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
HINT: Please make sure that the header name is correct. Also please check if you've specified all component dependencies with 'idf_component_register(REQUIRES ...)'. If the component is not present then it should be added by the IDF Component Manager. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
```

CMakeLists.txt (/):
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.5)

if(NOT DEFINED ENV{ESP_MATTER_PATH})
    message(FATAL_ERROR "Please set ESP_MATTER_PATH to the path of esp-matter repo")
endif(NOT DEFINED ENV{ESP_MATTER_PATH})

if(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})
    if("${IDF_TARGET}" STREQUAL "esp32" OR "${IDF_TARGET}" STREQUAL "")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32c3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c3_devkit_m)
    elseif("${IDF_TARGET}" STREQUAL "esp32c2")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c2_devkit_m)
    elseif("${IDF_TARGET}" STREQUAL "esp32h2")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32h2_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32s3")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32s3_devkit_c)
    elseif("${IDF_TARGET}" STREQUAL "esp32c6")
        set(ENV{ESP_MATTER_DEVICE_PATH} $ENV{ESP_MATTER_PATH}/device_hal/device/esp32c6_devkit_c)
    else()
        message(FATAL_ERROR "Unsupported IDF_TARGET")
    endif()
endif(NOT DEFINED ENV{ESP_MATTER_DEVICE_PATH})

set(PROJECT_VER "1.0")
set(PROJECT_VER_NUMBER 1)

set(ESP_MATTER_PATH $ENV{ESP_MATTER_PATH})
set(MATTER_SDK_PATH ${ESP_MATTER_PATH}/connectedhomeip/connectedhomeip)

# This should be done before using the IDF_TARGET variable.
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
include($ENV{ESP_MATTER_DEVICE_PATH}/esp_matter_device.cmake)

set(EXTRA_COMPONENT_DIRS
    "${ESP_MATTER_PATH}/examples/common"
    "${MATTER_SDK_PATH}/config/esp32/components"
    "${ESP_MATTER_PATH}/components"
    "${ESP_MATTER_PATH}/device_hal/device"

    "${CMAKE_CURRENT_LIST_DIR}/../../../libraries/esp-aws-iot/libraries"
    ${extra_components_dirs_append})

project(outlet)

# WARNING: This is just an example for using key for decrypting the encrypted OTA image
# Please do not use it as is.
if(CONFIG_ENABLE_ENCRYPTED_OTA)
    target_add_binary_data(light.elf "esp_image_encryption_key.pem" TEXT)
endif()

if(CONFIG_IDF_TARGET_ESP32C2)
    include(relinker)
endif()

idf_build_set_property(CXX_COMPILE_OPTIONS "-std=gnu++14;-Os;-DCHIP_HAVE_CONFIG_H" APPEND)
idf_build_set_property(C_COMPILE_OPTIONS "-Os" APPEND)
# For RISCV chips, project_include.cmake sets -Wno-format, but does not clear various
# flags that depend on -Wformat
idf_build_set_property(COMPILE_OPTIONS "-Wno-format-nonliteral;-Wno-format-security" APPEND)
```
CMakeLists.txt (/main/):
```
set(PRIV_REQUIRES_LIST device esp_matter esp_matter_console app_reset)

idf_component_register( SRCS                "app_main.cpp"
                                            "app_driver.cpp"

                                            "networking/mqtt/subscription_manager.c"
                                            "networking/mqtt/core_mqtt_agent_manager.c"
                                            "networking/mqtt/core_mqtt_agent_manager_events.c"

                        INCLUDE_DIRS        "."

                        SRC_DIRS            "."
                        PRIV_INCLUDE_DIRS   "."
                        PRIV_REQUIRES       ${PRIV_REQUIRES_LIST})

set_property(TARGET ${COMPONENT_LIB} PROPERTY CXX_STANDARD 14)
target_compile_options(${COMPONENT_LIB} PRIVATE "-DCHIP_HAVE_CONFIG_H")
```

The backoffAlgorithm lives under `esp-aws-iot/libraries/backoffAlgorithm` but seems like it is not being compiled during the `idf.py build` process.

shubhamdp, 2023-07-13:
@law-ko Backoff Algorithm is a submodule in esp-aws-iot, just to double check, have you recursively initialized submodules using `git submodule --init --recursive`?

Also, you have to specify the component dependencies in `REQUIRES` or `PRIV_REQUIRES`  as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#when-writing-a-component) when registering the component using `idf_component_register`. If you are using the backoff algorithm in main component then I do not see `backoffAlgorithm` added as the dependency when registering component.

law-ko, 2023-07-21:
Solved by adding the REQUIRES section in CMakeLists.txt in main

closed 2023-07-21 by law-ko