espressif/esp-matter #511: Error when installing esp-matter - python setup.py bdist_wheel did not run successfully

ydanrubinstein, 2023-07-13:
I've installed esp-idf successfully and can build/flash and monitor the blink example on the ESP32H2 without issue.

However, when I try to install esp-matter following the [instructions here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#host-setup), installation fails at the ./install.sh stage as below (and it also looks like it has issues with pigweed (pw) packages - is it supposed to skip setting them up? Environment checks do pass).

I am on a Mac M1 Max (Ventura 13.4.1). 

Welcome any guidance or suggestions. Thanks!

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (8.9s)
  Setting up Python environment.....done (44.9s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20230713 00:41:22 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [662 lines of output]

Full session log here:
[esp-matter installation - Terminal Saved Output.txt](https://github.com/espressif/esp-matter/files/12036767/esp-matter.installation.-.Terminal.Saved.Output.txt)

jonsmirl, 2023-07-14:
Build instructions...
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md

ydanrubinstein, 2023-07-14:
Hi Jon - thanks for sending that link. I have already gone through those build instructions as well. It seems to be failing at bootstrap.sh.

1) Can you confirm that it is normal behavior that pw (pigweed) package installation and environment variable setting should be skipped?

2) The subprocess python setup.py bdist_wheel is failing and it's unclear to me why and what I can try to remedy.

Below is the output of running bootstrap.sh. Thanks! Dan.

ydr@MacBook-Pro connectedhomeip % source scripts/bootstrap.sh
Synchronizing submodule url for 'examples/common/QRCode/repo'
Synchronizing submodule url for 'examples/common/m5stack-tft/repo'
Synchronizing submodule url for 'third_party/boringssl/repo/src'
Synchronizing submodule url for 'third_party/editline/repo'
Synchronizing submodule url for 'third_party/freertos/repo'
Synchronizing submodule url for 'third_party/jsoncpp/repo'
Synchronizing submodule url for 'third_party/libwebsockets/repo'
Synchronizing submodule url for 'third_party/mbedtls/repo'
Synchronizing submodule url for 'third_party/nanopb/repo'
Synchronizing submodule url for 'third_party/nlassert/repo'
Synchronizing submodule url for 'third_party/nlfaultinjection/repo'
Synchronizing submodule url for 'third_party/nlio/repo'
Synchronizing submodule url for 'third_party/nlunit-test/repo'
Synchronizing submodule url for 'third_party/openthread/repo'
Synchronizing submodule url for 'third_party/pigweed/repo'
Synchronizing submodule url for 'third_party/pybind11/repo'

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (3.4s)
  Setting up Python environment.....done (3.8s)
  Setting up pw packages............skipped (0.1s)
  Setting up Host tools.............done (0.1s)

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

20230714 13:37:09 INF Environment passes all checks!

Environment looks good, you are ready to go!

To reactivate this environment in the future, run this in your 
terminal:

  source ./activate.sh

To deactivate this environment, run this:

  deactivate

Installing pip requirements for all...
  error: subprocess-exited-with-error
[... 646 lines cut ...]
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/ecdsa_secp224r1_sha3_256_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/ecdsa_secp224r1_sha256_p1363_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_signature_3072_sha512_256_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_signature_2048_sha384_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_pss_3072_sha256_mgf1_32_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_signature_3072_sha3_512_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_signature_3072_sha256_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/ecdsa_secp224r1_sha224_p1363_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      copying lib/Crypto/SelfTest/Signature/test_vectors/wycheproof/rsa_signature_2048_sha3_512_test.json -> build/lib.macosx-10.9-universal2-cpython-39/Crypto/SelfTest/Signature/test_vectors/wycheproof
      running build_ext
      building 'Crypto.Hash._MD2' extension
      creating build/temp.macosx-10.9-universal2-cpython-39
      creating build/temp.macosx-10.9-universal2-cpython-39/src
      clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -iwithsysroot/System/Library/Frameworks/System.framework/PrivateHeaders -iwithsysroot/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/Headers -arch arm64 -arch x86_64 -Werror=implicit-function-declaration -Wno-error=unreachable-code -DPYCRYPTO_LITTLE_ENDIAN -DSYS_BITS=64 -DLTC_NO_ASM -Isrc/ -I/Users/ydr/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/pigweed-venv/include -I/Appl [... cut ...]
         34 | #include <stdio.h>
            |          ^~~~~~~~~
      1 error generated.
      error: command '/Users/ydr/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/bin/clang' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects

jonsmirl, 2023-07-15:
https://github.com/project-chip/connectedhomeip/issues/24855

shubhamdp, 2023-08-10:
@ydanrubinstein Is this still an issue?

dhrishi, 2023-08-28:
Closing this now. Please re-open if you still face issues

closed 2023-08-28 by dhrishi