espressif/esp-matter #513: can't make apple home to use rgb

MgTheProgrammer, 2023-07-13:
I'll start by saying that my problem is mostly caused by the fact I am clueless in building diy iot devices and iv'e struggled in finding a beginners guide to matter (and specifically esp). 

To get my feet wet I've decided to make the simplest project I could have think of, rgb light. I have builded the light example from esp-matter and got a light that I can control it's temperature, I've looked at the source code and figured that changing 
```     
light_config.color_control.color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
light_config.color_control.enhanced_color_mode = EMBER_ZCL_COLOR_MODE_COLOR_TEMPERATURE;
``` 
might change the color wheel in home app to an rgb wheel from temperature wheel.
so I've changed it to:
```
    light_config.color_control.color_mode = EMBER_ZCL_COLOR_MODE_CURRENT_HUE_AND_CURRENT_SATURATION;
    light_config.color_control.enhanced_color_mode = EMBER_ZCL_COLOR_MODE_CURRENT_HUE_AND_CURRENT_SATURATION;
```
which I've found somewhere in the src code.
it didn't change anything in the home app.

p.s. currently im focused on making the home app showing this as an rgb light so I haven't tried to find how to control the gpio, but I can't find where it is even defined which port it uses. 

also not really matter related but what can I do to make the home app forget the devices I connected before, when building the device after iv'e connected it previosly the app recognize it as two matter devices. taking another apple device seems to resolve that issue so it must be cache related

jonsmirl, 2023-07-14:
Hue (HS) support is optional, ColorXY (XY) is mandatory. Do you have ColorXY implemented?

[image]

PSONALl, 2023-07-19:
@MgTheProgrammer The ColorMode attribute indicates which attributes are currently determining the color of the device.

For extended color light, since HS feature is optional,and it is not implemented in the SDK, you can't set color mode to Hue-Saturation, You would need to enable the feature Hue-saturation first extended color light to get the RGB light working
e.g. you can add it to the example like this:
```
    cluster_t *cluster = cluster::get(endpoint, ColorControl::Id);
    cluster::color_control::feature::hue_saturation::config_t hue_saturation_config;
    hue_saturation_config.current_hue = DEFAULT_HUE;
    hue_saturation_config.current_saturation = DEFAULT_SATURATION;
    cluster::color_control::feature::hue_saturation::add(cluster, &hue_saturation_config);
```
For setting your GPIOs you can check the driver code [here](https://github.com/espressif/esp-matter/blob/main/device_hal/led_driver/gpio/led_driver.c) 
For removing the device from apple home, you can click on the device and then click  "Remove Accessory" option under Settings

shubhamdp, 2023-08-02:
@MgTheProgrammer any updated on this?

MgTheProgrammer, 2023-08-02:
@shubhamdp sadly I have been way too busy to get to play with it. @jonsmirl @PSONALl thanks for taking the time helping me, I promise I'll give it a try as soon as I'll be able to.

dhrishi, 2023-08-12:
@MgTheProgrammer I am closing the issue now. Feel free to re-open this if you face any problems

closed 2023-08-12 by dhrishi