espressif/esp-matter #514: Using chip-tool to commission a device with DAC created using mfg_tool

thisal29, 2023-07-13:
Hi,

I have created a DAC and other certificates using mfg_tool. Following was the command that I used. 

`./mfg_tool.py -cn "My bulb" -v 0xFFF2 -p 0x8001 --pai \
   -k ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
   -c ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
   -cd ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der`

Afterwards I flashed the firmware. Then flahsed the binary image to the device using esptool.py using the correct address of the partition table. 

`esptool.py -p /dev/ttyUSB0 write_flash 0x10000 /home/thisal/esp/esp-matter/tools/mfg_tool/out/fff2_8001/35c350b3-febc-4dcd-9532-63746675447c/35c350b3-febc-4dcd-9532-63746675447c-partition.bin 0x17000 build/ota_data_initial.bin 0x29d000 build/storage.bin
`

After rebooting, the credentials were updated properly, the device was not using the Example DAC, but was responding to the custom DAC that was created. 

Next, I tried to commission the device using chip-tool. 

` ./chip-tool pairing ble-wifi <node> <wifi ssid> <password> <passphrase> <discriminator> --paa-trust-store-path ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/.
`

However, the ship-tool fails. Here is the console on the chip-tool. The full log is attached. 

...................................................................................................................
[1689258282.243186][126581:126583] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0001
[1689258282.243193][126581:126583] CHIP:CTL: Received Attestation Information from the device
[1689258282.243199][126581:126583] CHIP:CTL: Successfully finished commissioning step 'SendAttestationRequest'
[1689258282.243203][126581:126583] CHIP:CTL: AutoCommissioner setting attestationElements buffer size 278/278
[1689258282.243213][126581:126583] CHIP:CTL: Commissioning stage next step: 'SendAttestationRequest' -> 'AttestationVerification'
[1689258282.243216][126581:126583] CHIP:CTL: Performing next commissioning step 'AttestationVerification'
[1689258282.243218][126581:126583] CHIP:CTL: Verifying attestation
[1689258282.250586][126581:126583] CHIP:CTL: Failed in verifying 'Attestation Information' command received from the device: err 604. Look at AttestationVerificationResult enum to understand the errors
[1689258282.250604][126581:126583] CHIP:CTL: Error on commissioning step 'AttestationVerification': 'src/controller/CHIPDeviceController.cpp:1111: CHIP Error 0x000000AC: Internal error'
[1689258282.250606][126581:126583] CHIP:CTL: Failed to perform commissioning step 8
[1689258282.250608][126581:126583] CHIP:CTL: Going from commissioning step 'AttestationVerification' with lastErr = 'src/controller/CHIPDeviceController.cpp:1111: CHIP Error 0x000000AC: Internal error' -> 'Cleanup'

............................................................................................................................................

The error 604  is AttestationVerificationResult enum;
kCertificationDeclarationInvalidVendorId    = 604,

I was following which is similar.
https://github.com/project-chip/connectedhomeip/issues/26142

Any input regarding how to use certificates generated with mfg_tool along with chip-tool would be highly appreciated. 

Thanks and regards,
Thisal

   
[chip-tool_log.txt](https://github.com/espressif/esp-matter/files/12040715/chip-tool_log.txt)
[device_console_log.txt](https://github.com/espressif/esp-matter/files/12040716/device_console_log.txt)

jonsmirl, 2023-07-14:
Your vendor ID is menuconfig is likely set to 0xFFF1 and you used the 0xFFF2 certificates. Change it to 0xFFF2 in menuconfig.

thisal29, 2023-07-14:
@jonsmirl Thank you very much. 

I had to change both vendor_id and product_id to match with the arguments of the certificates. And it worked perferctly!

CONFIG_DEVICE_VENDOR_ID=0xFFF2
CONFIG_DEVICE_PRODUCT_ID=0x8001

closed 2023-07-14 by thisal29