espressif/esp-matter #515: Data model without ESP-Matter

Gonta01, 2023-07-14:
There are examples of esp32 device in connectedhomeip repository (for example, lighting-app). Also there is an example zap-light in esp-matter repository. It isn't clear how Data model is configured in these cases. 
Are there any information about that or can someone explain?

wqx6, 2023-07-14:
For examples except zap-light example in esp-matter, the date model is dynamically generated when running the examples. We will call `node::create()` and `endpoint::create()` to create the data model. And for zap-light and all the examples in connectedhomeip repository, the data model is statically generated by ZAP when building the examples. we will use `chip_app_component_codegen` and `chip_app_component_zapgen` to generate the data model codes in [CMake file](https://github.com/espressif/esp-matter/blob/main/examples/zap_light/main/CMakeLists.txt#L10-L11).

Gonta01, 2023-07-14:
Thank you for answer. How to create own data model using ZAP? Should files .matter and .zap somehow changed?

wqx6, 2023-07-14:
Yes, you should change the `.matter` and `.zap` files to create own data model.

Gonta01, 2023-07-14:
They cannot be modified manually. Right? ZAP tool is used to generate `.zap`. What tool is used to generate `.matter`?
Are there anywhere a step-by-step guide?

wqx6, 2023-07-14:
https://github.com/project-chip/connectedhomeip/blob/master/docs/code_generation.md Here is the code-generation doc.

Gonta01, 2023-07-14:
@wqx6 thank you!

closed 2023-07-14 by Gonta01