espressif/esp-matter #516: Setup of esp-matter

SYESHASWINI, 2023-07-14:
Hello, I unfortunately messed up the esp-matter directory in the environment, when I try to reinstall it gets the latest version where my programs ends up with errors. So I which to download the old version of it. Can I get the exact branch to clone it. 
Thanks.

dhrishi, 2023-07-15:
@SYESHASWINI You can reset to the earlier commit id that was working and update the submodules. Another option is to make changes to your program to support the latest codebase

SYESHASWINI, 2023-07-15:
@dhrishi I want get the tree branch [release/v1.0](https://github.com/espressif/esp-matter/tree/release/v1.0) of esp-matter repo, but it says like 
**fatal: could not create work tree dir 'esp-matter': File exists**.
Tried with commit id as well didn't work.

shubhamdp, 2023-07-16:
@SYESHASWINI Please try at the different directory location. Or If you want to keep using the previous clone then you can do the following...
```
cd esp-matter
rm -rf connectedhomeip/connectedhomeip/.environment
git fetch origin release/v1.0

# NOTE: If you have any modified files, you will have to either discard them or stash them before checkout.
git checkout origin release/v1.0
./install.sh
```

SYESHASWINI, 2023-07-17:
@shubhamdp when I execute git fetch origin matter_1_1/te1 command
it shows the error as
**error: pathspec 'matter_1_1/te1' did not match any file(s) known to git**

shubhamdp, 2023-07-17:
I'd say then just run `git fetch` and checkout desired branch.

We have:
- main: For Matter v1.2, (We add the newer features here)
- release/v1.1 - For Matter v1.1
- release/v1.0 - For Matter v1.0

SYESHASWINI, 2023-07-17:
- @shubhamdp while executing git fetch origin release/v1.0 there no error, then at git checkout origin release/v1.0 command it throws the error.
- In the release/v1.0 we have this code snippet for fabric removal where as in the previous version it is not available in the initial version of matter.
 **case chip::DeviceLayer::DeviceEventType::kFabricRemoved:**

shubhamdp, 2023-07-17:
Sorry, my bad `git checkout origin release/v1.0 `, ditch the origin from the name please, `git checkout release/v1.0`.

SYESHASWINI, 2023-07-17:
@shubhamdp I tried cloning the tree branch using the command.
git clone -b <branch name> https://github.com/espressif/esp-matter.git 
where the connectedhomeip directory is empty in this process.

Later executed the below command:
cd esp-matter
rm -rf connectedhomeip/connectedhomeip/.environment
git fetch origin release/v1.0
git checkout  release/v1.0
which says Your branch is up to date.
How do I figure it out

shubhamdp, 2023-07-17:
Please go through this doc https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#getting-the-repositories

SYESHASWINI, 2023-07-17:
@shubhamdp tried this which gives the latest version , while building the previous projects throws error with this. To get the specified branch from the repo we don't have the branch path.
[image]

shubhamdp, 2023-07-17:
@SYESHASWINI You can use `--single-branch <branch/tag>` option in your clone command to clone the specific branch.

SYESHASWINI, 2023-07-17:
@shubhamdp Thank You.
In the release/v1.0 - For Matter v1.0 we have code snippet for fabric remove for previous version it wasn't there. Is there any version before this.

SYESHASWINI, 2023-07-18:
@shubhamdp I have cloned the matter_1_1/te1 branch of esp-matter repo, when I try to build my previous project I came across the error as:
CMake Error at /home/ubuntu/esp/esp-idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'route_hook'.
Call Stack (most recent call first):
  /home/ubuntu/esp/esp-idf/tools/cmake/build.cmake:232 (__build_resolve_and_add_req)
  /home/ubuntu/esp/esp-idf/tools/cmake/build.cmake:507 (__build_expand_requirements)
  /home/ubuntu/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:41 (project)

Is there anything that I need to change.

closed 2023-07-19 by SYESHASWINI