espressif/esp-matter #518: E (1080) esp_matter_feature: Cluster shall support Setback feature

albrownwood, 2023-07-16:
Getting the following error in log 
`E (1080) esp_matter_feature: Cluster shall support Setback feature`

the thermostat setback feature is added as follow:
```
   //setback
   cluster::thermostat::feature::setback::config_t setback_config;
   setback_config.occupied_setback = 20;
   setback_config.occupied_setback_max = 100;
   setback_config.occupied_setback_min = 10;
   cluster::thermostat::feature::setback::add(thermostat_cluster, &setback_config);
```

shubhamdp, 2023-07-17:
@albrownwood Are you also adding the "occupancy" feature. I see this error when adding [occupancy feature](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_feature.cpp#L742).

In specification, the occupancy is not by itself, and always accompanied by either, heating, cooling, or setback feature and hence code is checks for those bits.

I would suggest you add the setback first and them occupancy and it would work as expected.

And I'll try to get the documentation fixed for this.

albrownwood, 2023-07-17:
@shubhamdp indeed adding the occupancy features after heating, cooling, or setback solves it.
I'll let you decide if you want to close this issue or keep it open to track a documentation fix.

Thanks!

shubhamdp, 2023-07-18:
@albrownwood Please keep this open, will close this with the commit.

closed 2023-08-24 by espressif-bot