espressif/esp-matter #519: How to udate Device specific info in Basic Info Cluster in firmware ?

aggaddam, 2023-07-17:
I want to update device-specific info in the Basic Info cluster in firmware. Currently, It is updated through the SDK configuration using the menu config. Is it possible we can update the below information in firmware without using Menuconfig as we are storing the below information in firmware during the device provisioning process?

[image]

jonsmirl, 2023-07-17:
https://github.com/espressif/esp-matter/tree/main/tools/mfg_tool

aggaddam, 2023-07-18:
@jonsmirl  Thanks for the link. 

I want to directly update the mfg-specific data received on the UART port in our firmware to NVS instead of generating the mfg factory partition. Is it possible with ESP -Matter SDK?

aggaddam, 2023-07-18:
@shubhamdp Thanks for providing the information. I think this will help me to update the configurations that the chip SDK needs from firmware.I will explore more on this

jonsmirl, 2023-07-18:
Just write the stuff into the NVS partition yourself using IDF APIs. Look at mfg_tool for what the data looks like. 
Scroll down in this issue and you can see a dump of a partition.
https://github.com/espressif/esp-matter/issues/316

shubhamdp, 2023-07-19:
@aggaddam Both tools do the same work with some differences.

1. mfg_tool.py -> generates the partitions along with onboarding codes. Also, this was targeted for mass manufacturing, and can generate more than one partition. It also generates the artefacts that can be used afterwards.

2. generate_esp32_chip_factory_bin.py -> This only generates one partition and was targeted for testing manufacturing data, and we wanted to have some tool in connectedhomeip repository.

I would suggest you use `mfg_tool.py`  as it generates the our artefacts that can be used later and pretty aligned with Espressif's manufacturing process.

shubhamdp, 2023-07-20:
@aggaddam If your queries are answered can you please close the issue?

dhrishi, 2023-07-20:
Closing this now