espressif/esp-matter #520: thermostat: occupied_setback initial value does not seem to be registered

albrownwood, 2023-07-17:
I initially add the occupied_setback initial config as follow:

```
    //setback
    cluster::thermostat::feature::setback::config_t setback_config;
    setback_config.occupied_setback = 20;
    setback_config.occupied_setback_max = 100;
    setback_config.occupied_setback_min = 10;
    cluster::thermostat::feature::setback::add(thermostat_cluster, &setback_config);
```

Then later in the PRE-UPDATE callback, I try to read that value:
```
        value = esp_matter_invalid(NULL);
        attribute = attribute::get(cluster, Thermostat::Attributes::OccupiedSetback::Id);
        esp_err_t error = attribute::get_val(attribute, &value);
        if(error){
            ESP_LOGW(TAG, "unable to get_val OccupiedSetback with error %s", esp_err_to_name(error));
        }
        uint8_t occupied_setback = value.val.u8;

        ESP_LOGW(TAG, "to debug: occupied_setback %d", occupied_setback);
```

The output in the debug console is always the following no matter the initial value:
```
I (69992) app_main: PRE_UPDATE
W (70002) app_main: to debug: occupied_setback 255
```

shubhamdp, 2023-07-19:
@albrownwood Can you please share the patch, so that I can give a try.

jadhavrohit924, 2023-07-19:
@albrownwood I tested the issue and I was able to get correct value for `OccupiedSetback`. Please share your patch.
```
W (2502) app_main: to debug: occupied_setback 20
```

albrownwood, 2023-07-19:
while preparing the minimal example, I figured out the issue was on my side. I probably forgot to do an erase-flash and commission again after adding the setback feature in code and test it. Thanks for the help and for double checking the feature is working fine, this can be closed.

closed 2023-07-19 by albrownwood