espressif/esp-matter #521: Failed to use factory data as commisioning data provider

eme-bennytang, 2023-07-18:
We are trying to use the ESP32 factory partition for the data for 
- Commisionable data
- Device Attestation Data
- Device instance Info

I have copied light_switch in the examples, and added below config in the sdkconfig.defaults

**CONFIG_ENABLE_CHIP_SHELL=n
CONFIG_BUILD_CHIP_TESTS=n
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_CONFIG_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_COUNTERS_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_CHIP_KVS_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_FACTORY_COMMISSIONABLE_DATA_PROVIDER=y
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y**

Below is the generated onboard code excel
[image] 

But somehow when I flashed the project and flashed generated partition data, the device seems to fail to get pass code, but isntead used the default pass code. 
And as a result, my QR code/Manual pairing code are different to the generated ones.
[image]

Attached the log from the monitor
[light_switch_log.txt](https://github.com/espressif/esp-matter/files/12077660/light_switch_log.txt)

ESP IDF version: v5.0.2
Matter SDK: release/v1.1 548ef58b604ef1df208e4a040803a4218e213103

eme-bennytang, 2023-07-19:
More info:
I actually tried to pair with the QR code/Manual code displayed in the monitor log, but failed.
But with the excel QR code/Manual code, the commissioning succeeded.

So I think there is a bug there

eme-bennytang, 2023-07-19:
Hi!
We have the need to extract the QR code / Manual Pairing code to display on screen
[image]
I used the above API to retrieve the QR code/Manual Code.

The problem is the extracted QR code/manual code from program isnt matching whats programmed (in fctry partition) and whats effective (the fctry partition one is effective by testing commissioning)

So its definitely an issue here unless I am using the wrong code to extract those information.

dhrishi, 2023-07-19:
@eme-bennytang The device just has the salt and verifier and not the actual passcode. Hence, it is not able to generate the QR code or manual pairing code that has the passcode. The APIs above should be only for test/development and would be using the test values (say 20202021) directly.

eme-bennytang, 2023-07-19:
I am not sure if I understand this correctly, please correct me if I am wrong:
- Although generated factory partition has --passcode option (or random generated if not input), but the passcode is just a numeric representation of the "salt" and "Spake2p verifier", which both are actually stored in the partition, but not the passcode
- So when the program tried to read passcode from the partition it will fail since no actual "passcode" stored, just "salt" and "Spake2p verifier", but this 2 information will be combined to a effective passcode (which is the one in the generated partition summary file)
- With same concept, QR code is the ASCII representation of passcode/discriminator/vendorID/productID....etc, thus not really the effective one
- Thus explain why I can commission with the summary QR code/manual code

My next question is: if we need really need to extract those (QR code/manual code) in code and display it on screen (for commissioning), what better way can we do that?
e.g. Get the "salt" and "verifier" and do the same numerical operation to generate the result passcode and hence QR code/Manual code?

dhrishi, 2023-07-19:
@eme-bennytang 

>"salt" and "Spake2p verifier", which both are actually stored in the partition, but not the passcode

That's right.

>So when the program tried to read passcode from the partition it will fail since no actual "passcode" stored, just "salt" and "Spake2p verifier"

Correct.

>but this 2 information will be combined to a effective passcode (which is the one in the generated partition summary file)

It's the other way round. The salt and verifier are calculated from the passcode

>My next question is: if we need really need to extract those (QR code/manual code) in code and display it on screen (for commissioning), what better way can we do that?
e.g. Get the "salt" and "verifier" and do the same numerical operation to generate the result passcode and hence QR code/Manual code?

You cannot generate passcode from salt and verifier. Passcode to verifier is a one way operation. The only option is to to have the actual passcode on the device and then create the salt and verifier from it in the firmware.

eme-bennytang, 2023-07-20:
Thanks both for the clear explanation.

In fact we are displaying those info on screen AND print it in the package box.
Anyway, I think the only way is to extract those info while generating the factory partition and program it to some region to read in code.

For the mfgtool.py, is it possible to also program the passcode with the key "pin-code" so that I can read it with the test API?
[image]

Otherwise I will just program it in other partition and read it with our own implementation

shubhamdp, 2023-07-20:
@eme-bennytang I would strongly suggest you to please read through Matter Specification Section 5.1.7. Generation of the Passcode before doing anything.

We did not add the passcode in the manufacturing partition due to security considerations listed in the above section.

What you see in the `ESP32Config.[cpp | h]` could be the leftover and I think it's better we don't keep it alongside the verifier and salt.

eme-bennytang, 2023-07-20:
@shubhamdp 
So are you saying if we are using static passcode, there is no way that we can read those information FROM the unit, but instead need to gather the information generated in the summary? 

How am I suppose to check in production stage whether my programmed static passcode is the one I am expecting?
There must be a way to check against the outside passcode against the passcode inside the unit?  (QR/Manual code in the summary vs the effective QR/Manual code in the unit to ensure the printed label is correct)

shubhamdp, 2023-07-20:
> @shubhamdp
So are you saying if we are using static passcode, there is no way that we can read those information FROM the unit, but instead need to gather the information generated in the summary?

@eme-bennytang I am not saying anything, I am stating what is in the specification. Please see the below excerpt from the specification.

> If the static passcode is supplied to the device, its storage location SHALL be physically isolated from the location where the PAKE verifier is stored and SHALL only be accessible through local interfaces and SHALL NOT be accessible to the executing unit handling the PAKE verifier.

 You can still have the passcode in the device but make sure it is isolated from where verifier is stored.

> How am I suppose to check in production stage whether my programmed static passcode is the one I am expecting?
There must be a way to check against the outside passcode against the passcode inside the unit? (QR/Manual code in the summary vs the effective QR/Manual code in the unit to ensure the printed label is correct)

You can try commissioning the device with the passcode/qrcode/manual-pairing-code and see if that works. Or, there are tools to generate the verifier from the passcode and you can validate the verifier. ([c++](https://github.com/project-chip/connectedhomeip/tree/master/src/tools/spake2p) and [python](https://github.com/project-chip/connectedhomeip/tree/master/scripts/tools/spake2p)).

Event better, just do a PASE with the pincode on the box with that device which contains the corresponding verifier and that way you can validate that it connected with the same device.

But, if you still want it in the device, then I would suggest please store it somewhere safe as per the constraints laid out by specifications.

PS: Pincode is part of qrcode and manual pairing code

eme-bennytang, 2023-07-20:
@shubhamdp Well understood, Thanks for the great help all! I will be closing the issue

closed 2023-07-20 by eme-bennytang