espressif/esp-matter #522: Device pairing Issue

SYESHASWINI, 2023-07-19:
I'm using 
esp-idf: v4.4.3
esp-matter: release/v1.0 
 I changed the bluetooth mode to bluedroid -dual mode. Then tried to add sample light example, it shows as failed. When I check the same example with Nimble it added successfully.
[pairing issue.txt](https://github.com/espressif/esp-matter/files/12091711/pairing.issue.txt)

SYESHASWINI, 2023-07-20:
Do I need to change any bluedroid configurations. It seems like the bluedroid is unable to initiate BLE.

sayondeep, 2023-07-21:
@SYESHASWINI 
Have you made any changes to the above branches you mentioned?

On using your sdkconfig on my set up...I see that you have disabled mbedtls_hkdf_c ..which is by default enabled on set-target command....that is causing the linking error.

Again linking error occurs when bluetooth v5.0 features is enabled under bluedroid options.

CONFIG_BT_BLE_50_FEATURES_SUPPORTED = n

CONFIG_MBEDTLS_HKDF_C = y

On making the above changes I am able to run the light example and commission it successfully.

SYESHASWINI, 2023-07-21:
@sayondeep Made the changes and tried commissioning, but it is failed.
[sdk_.txt](https://github.com/espressif/esp-matter/files/12116995/sdk_.txt)

shubhamdp, 2023-07-21:
@SYESHASWINI could be related to https://github.com/project-chip/connectedhomeip/pull/25308, can you please try to cherry-pick this change on release/v1.0 branch and try it again?

```
cd esp-matter/connectedhomeip/connectedhomeip
curl https://patch-diff.githubusercontent.com/raw/project-chip/connectedhomeip/pull/25308.patch | git am
```

SYESHASWINI, 2023-07-21:
@shubhamdp unable to patch this.
[image]

shubhamdp, 2023-07-21:
@SYESHASWINI see if you have any pending rebase/cherry-pick going on, stash/clean it up and then apply.

Or its a simple patch, open `connectedhomeip/connectedhomeip/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp` and change the line 655 (hopefully correct line number if you are on release/v1.0, or for reference [this](https://github.com/espressif/connectedhomeip/blob/4088a77f557e8571a39338fad51a1d8eb0131d79/src/platform/ESP32/bluedroid/BLEManagerImpl.cpp#L655) line).

From 
```
    advData[index++] = 0x0A;                            // length
```
to
```
    advData[index++] = 0x0B;                            // length
```

SYESHASWINI, 2023-07-21:
@shubhamdp after the change the devices are commissioning and adding successfully, but there is a inconsistency while adding. The devices are not adding for every attempt. Do we need to change any configuration.

closed 2023-07-21 by SYESHASWINI

Ajay2803, 2025-02-26:
Hello, @SYESHASWINI 

can you help me to do same, I am facing the same issue as you, but able to solve using this methods.