espressif/esp-matter #524: How do I obtain information about paired nodes

maenkai, 2023-07-20:
for example:
After commisin the device, I can get the specific information under each node, such as node 1 is homekit, node 2 is google

shubhamdp, 2023-07-20:
Where do you want to obtain these details, on the commissioned Node or on the controller?

On Node: you can iterator over the fabrics and get the VendorId, you can find the relevant APIs in [`FabricTable.h`](https://github.com/project-chip/connectedhomeip/blob/master/src/credentials/FabricTable.h).

On Controller: You can read the fabrics attribute in operational credentials cluster with fabric-filtered set to false. `chip-tool operationalcredentials read fabrics <node-id> 0 --fabric-filtered 0`

maenkai, 2023-07-20:
> Where do you want to obtain these details, on the commissioned Node or on the controller?
>
> On Node: you can iterator over the fabrics and get the VendorId, you can find the relevant APIs in [...]

Thanks you very much!

closed 2023-07-20 by maenkai