espressif/esp-matter #525: How to update manufacturer specific information in scan response in BLE advertisment for matter enabled device ?

aggaddam, 2023-07-20:
I want to use the ble scan response data for manufacturer-specific information. But, it appears that currently there is no way to set the scan response for matter BLE advertisement in ESP-Matter SDK. I checked in SDK configuration I didn't find any option for the scan response data.  

Has anyone tried using scan response in the esp-matter SDK ?

shubhamdp, 2023-07-25:
@aggaddam Right now, there is no direct provision for configuring scan response, will add some APIs for the same.

shubhamdp, 2023-08-10:
We have raised a PR here, will update the esp-matter once this gets merged. https://github.com/project-chip/connectedhomeip/pull/28571

dhrishi, 2023-09-30:
@aggaddam Closing this as the scan response API is available in the latest esp-matter main branch

closed 2023-09-30 by dhrishi